apache/mynewt-nimble #25: IPv6 support over Bluetooth

sterlinghughes, 2018-04-03:
Both over 4.2 as well as directly over extended advertising framework.

Citrullin, 2018-12-07:
I am not that deep into this topic, but I would say, that's nothing which should be part of this library. You should consider using [myNewt](https://mynewt.apache.org/) or [RIOT OS](https://www.riot-os.org/) for this. Both are using this library. So you are able to use the 6Lowpan stack over bluetooth with it.

haukepetersen, 2019-11-12:
Just saw this issue again: preliminary IPv6-over-BLE support using NimBLE was merged into RIOT master now.

I think IPv6 over extended advertising framework is not even drafted by standardization bodies, right?!

Citrullin, 2019-11-13:
@haukepetersen I can confirm this. Already using it with the newest linux kernel. So, this issue not present anymore. Works with Bluetooth >= 4.0.

KKopyscinski, 2021-05-20:
I'm closing this, NimBLE Bluetooth stack can be used over 6LoWPAN implemented in OS that it's used on

closed 2021-05-20 by KKopyscinski