apache/mynewt-nimble #44: Extended Advertisement Issues

mwilliford, 2018-04-15:
Greetings,

I'm not 100% sure I'm using it correctly, but I am attempting to send 200byte (or larger I hope), packets over ble_ext_adv using phy-1m (with a goal to test phy-2m on secondary).  I have a test project which shows the following issues, and allows for a bit of stress testing against this part of nimble.

Test Project:
https://github.com/mwilliford/mynewt-extadv-test

Issues:
- Initially, lots of asserts.
- Apply diff to nimble, maybe a little better, heavy corruption though.
- Issue with scanning stopping after 5-10min. I see no BLE_GAP_EVENT_DISC_COMPLETE when it 
  stops. I know it stops because r resetting the 'other' board does not fix it. Only resetting the board 
  which stops decoding messages fixes it.

NOTE:  There is a file called nimble.diff which is an attempt to work around asserts, but it likely leaves the stack in a bad state, as the above issues might be related to a bigger state issue.

andrzej-kaczmarek, 2018-04-16:
The issue in controller triggered by this app is due to broken handling of `Duration` parameter in `LE Set Extended Advertising Enable`. Basically we allow aux to finish past the duration limit, which we should not do. If this aux started before time limit, we will (incorrectly) try to reschedule advertising event with invalid anchor which ultimately leads to an assert. I'll prepare fix.

Btw, the application divides `duration` by 10 two times: when sending an event to adv task and then again when enabling advertising set. This means actual duration passed to controller is 10 times less that expected.

mwilliford, 2018-04-16:
Thanks, should I use max_events instead for now as a workaround, would that be a more stable way to define the end of the broadcast?  

I will fix the divide by 10 issue, thanks.

andrzej-kaczmarek, 2018-04-16:
Yes, `max_events` should work fine.

mwilliford, 2018-04-16:
@andrzej-kaczmarek Were you able to repro it no longer receiving ext_adv messages after 5min or so?

andrzej-kaczmarek, 2018-04-16:
I did not try to check scanner yet, only the advertiser part which needs some fixing.

mwilliford, 2018-04-20:
Some more details:
-  I may have fixed the scanner issue, I am not sure I provided the right values to ble_gap_ext_disc in my first test cases.
- The asserts are fixed, confirmed that I no longer need my patch.

Issues which I am still hitting:
- As we briefly discussed, there is some issue with duration, max_events is testing way better.  However, when using either, I still hit situations where the callback registered with ble_gap_ext_adv_configure is not being called back.

At least one of those times, I caught it here:
   } else if (ble_ll_adv_active_chanset_is_sec(advsm)) {
        os_eventq_put(&g_ble_ll_data.ll_evq, &advsm->adv_sec_txdone_ev); // the event sometimes is already queued, bug?
    } else {
adv_sec_txdone_ev seems to already have been previously queued, thus os_eventq_put won't queue it again, which I think is at least one of the failure cases which results in no callback.
Because in my use-case I'm sending a stream of data, I need the callback to work so I know when to send the next set of data (and to clean up some memory).  

Another issue that I think we have, it seems that I can't send a aux-chain beyond 1 aux packet.  Meaning, I seem to be limited to around the 254 bytes of adv_data that fits into a single aux packet.  I'll try to gather more data on this. 

Thanks again for your help!

andrzej-kaczmarek, 2018-04-20:
@mwilliford 
we already identified issue with extended scanner, could be that you simply no longer trigger it with different parameters. I have fix for this, should be pushed to github today.

as for extended advertising could you please retry with latest master? I merged a lot of fixes for extended advertising there. also I had no problems sending 1650 bytes of data using your sample app by just changing appropriate #define.

if you still experience issues with some specific set of parameters, please update sample app so I can try to reproduce these with my devkit.

andrzej-kaczmarek, 2018-04-20:
fix for extended scanning: https://github.com/apache/mynewt-nimble/pull/57

mwilliford, 2018-04-20:
@andrzej-kaczmarek 
I have pushed interval fixes on my test project side as well.  I had some bad values which led to the scanner issue on the test project side as well.

However, with latest core and nimble, (as of master right now).  When I increase the packet size to 600 for example, actual packets received is less:

078643 Extended adv: completed rssi=-37 txpower=127, pphy=1, sphy=0, sid=0, addr_type=255 addr=00:00:00:00:00:00
078646 ********* NO SIZE MATCH - BAD!  expected = 600 : was = 110

So, it seems the 'other' side only received 110 when 600 should have been transmitted.  I don't know if the issue is on the receive side or tx side (I don't have a sniffer that can follow adv_ext aux)  I am testing this on nrf52832's, what hw are you testing on?  Does the test program pass for you?

Also, I will change the test program to match my code more closely, as it blindly transmits every 500ms, I will alter it to advertise next test packet after receiving callback.  This will likely surface the other bug I had reported before.

Marcus

mwilliford, 2018-04-20:
@andrzej-kaczmarek 

Ok, pushed updates to test project.  Cleaned up a little, and made max_event cause the adv stop, to help work around any race conditions (per your suggestion).

Test plan:
https://github.com/mwilliford/mynewt-extadv-test
- Run on two nrf52832's
- Notice max_events should happen prior to duration, so adv callback should happen on max_events trigger in ble_ll_adv.c
- Watch for bad packet (length incorrect) on receiver side.  
- Watch for advertising to stop, because callback never happens.  This takes between 1min to 5min usually.  This will happen even when device is not seeing any traffic from the other.

Tested against:
core: e897020434e50a490fcae29e1c6ed9c480c4b74e
nimble: 637883a5121dd9df8fc6f6e53a78de8698e0a83f

I'll keep testing to try to see if the issue is with the test program.  I the following large buffers setup, so a 600 byte packet should be no issue.
    BLE_HCI_EVT_BUF_SIZE: 1024
    BLE_EXT_ADV_MAX_SIZE: 1024

With these settings, the max_events 'shoul' happen before the duration timer hits, so this should avoid our concerns about duration timeout race conditions for now.
  EXTADV_DURATION:
    description: "duration of broadcast in ms"
    value: 600
  EXTADV_INTERVAL:
    description: "advertisement interval"
    value: 30
  EXTADV_MAX_EVENTS:
    description: "max events for broadcast"
    value: 6

Please let me know if you have issues with repro.

sjanc, 2018-04-23:
Hi,

There seems to be some misunderstanding about on how extended advertising scan works.

BLE_HCI_EVT_BUF_SIZE defines HCI event size taken from pool, by definition it cannot be larger than 257 (which is probably best value if you have extended scan enabled). 

When you do ext scan and remote is sending big data you'll get fragmented data from nimble and you need to take care of this in your application. This is to avoid caching in stack and keep it simple (at least for time being). data_status field will tell you if more data is expected to come (BLE_HCI_ADV_DATA_STATUS_COMPLETE, BLE_HCI_ADV_DATA_STATUS_INCOMPLETE, BLE_HCI_ADV_DATA_STATUS_TRUNCATED ).

Also, depending on BLE_HCI_EVT_BUF_SIZE or advertiser implementation, you may get data fragmented in any way, so you should make no assumptions about this.

Also, could you please confirm if original issues reported here are fixed?

mwilliford, 2018-04-23:
@sjanc 
I think you are correct that my test code does not properly reassemble, however I never get anything that is marked INCOMPLETE, all lines in my rx side observer parse it as completed.  Does this mean that the tx side is not doing the right thing?

case BLE_HCI_ADV_DATA_STATUS_COMPLETE:
            console_printf("completed");
            break;

008528 Extended adv: **completed** rssi=-41 txpower=127, pphy=1, sphy=0, sid=0, addr_type=255 addr=00:00:00:00:00:00
008530 ********* NO SIZE MATCH - BAD!  expected = 600 : was = 110

So, I can and will adjust my code to reassemble properly, but I am not getting any data_status which would be appropriate to reassemble.

mwilliford, 2018-04-24:
@sjanc 

It seems that a patch from @andrzej-kaczmarek helped me read data-status.
I have also updated my test code to no longer say "BAD" because @sjanc points out that it is my responsibility to re-assemble the packet on the receive side, so I should build a buffer and wait for complete.  

Here are the things I did:
- Reduced the test size to 256, so each byte is unique and easy to read, it should require 2 aux pdu's.
- Reduced BLE_HCI_EVT_BUF_SIZE to 254, per sjanc recommendation, higher is not required.
- Also ran test at 500Bytes tx and rx with full HCI debug turned on.  Attached file.
- I tested with and without filter_duplicates flag to see if that would de-dupe based on DID.

Results:
with 256byte test:
- I often see a packet of aux of 226, then aux of 16 bytes (showing bytes 0x00 = 0xf1), 242 bytes of my 256 byte test pattern.  Looks like the aux follows the next aux path to a "next" of 16 bytes
- I would think it should fragment based on 226, 30, and fit the adv_ext sequence in 2 aux frames.
- I don't see DID part of the AID, so I only see SID.  I am not sure how I can re-assemble host side without DID.  I understand that I can set SID to differ between sets, but shouldn't controller handle de-duplication based on adv.  OH, maybe I need to turn that on?

with 500 Byte test:
- Similar results, test pattern harder to confirm as it wraps around, but I am pretty sure I don't receive all data.
- I also see a value 0x4a in the rawheader of the hci on rx for secondaryphy, which does not seem to be right in the spec. (see attached file).
002910 [ts=22734328ssb, mod=4 level=0] rawhdr =01 20 00 01 b5 13 dc 28 91 cc 00 4a 04 7f de 00 00 00 00 00 00 00 00 00 10

----------------  Patch applied

diff --git a/nimble/host/src/ble_hs_hci_evt.c b/nimble/host/src/ble_hs_hci_evt.c
index 6125084..156fc5c 100644
--- a/nimble/host/src/ble_hs_hci_evt.c
+++ b/nimble/host/src/ble_hs_hci_evt.c
@@ -578,7 +578,7 @@ ble_hs_hci_evt_le_ext_adv_rpt(uint8_t subevent, uint8_t *data, int len)

     params = &ext_adv->params[0];
     for (i = 0; i < num_reports; i++) {
-        desc.props = (params->evt_type) & 0xFF;
+        desc.props = (params->evt_type) & 0x1F;
         if (desc.props & BLE_HCI_ADV_LEGACY_MASK) {
             legacy_event_type = ble_hs_hci_decode_legacy_type(params->evt_type);
             if (legacy_event_type < 0) {
@@ -587,7 +587,7 @@ ble_hs_hci_evt_le_ext_adv_rpt(uint8_t subevent, uint8_t *data, int len)
             }
             desc.legacy_event_type = legacy_event_type;
         } else {
-            desc.data_status = params->evt_type >> 8;
+            desc.data_status = params->evt_type >> 5;
         }
         desc.addr.type = params->addr_type;
         memcpy(desc.addr.val, params->addr, 6);

--------- debug info patch

diff --git a/nimble/host/src/ble_hs_dbg.c b/nimble/host/src/ble_hs_dbg.c
index 646f66f..c256425 100644
--- a/nimble/host/src/ble_hs_dbg.c
+++ b/nimble/host/src/ble_hs_dbg.c
@@ -149,7 +149,24 @@ ble_hs_dbg_le_event_disp(uint8_t subev, uint8_t len, uint8_t *evdata)
             BLE_HS_LOG(DEBUG, "PHY update. handle=%u tx=%u rx=%u\n",
                        get_le16(evdata + 1), evdata[3], evdata[4]);
         break;
+        case BLE_HCI_LE_SUBEV_EXT_ADV_RPT:
+            advlen = evdata[9];
+            rssi = evdata[10 + advlen];
+            BLE_HS_LOG(DEBUG, "\n\nLE EXT ADV Report. len=%u num=%u evtype=%x datastatus=%02x ",
+                    len, evdata[0], get_le16(evdata+1), evdata[1] >> 5
+                 );
+            BLE_HS_LOG(DEBUG,"addrtype=%u addr=%x.%x.%x.%x.%x.%x pphy=%u sphy=%u sid=%u txpow=%u rssi=%u advlen=%u\n",
+                       evdata[3], evdata[9], evdata[8], evdata[7], evdata[6],
+                       evdata[5], evdata[4],
+                       evdata[10], evdata[11], evdata[12], evdata[13], evdata[14], evdata[24] );
+            BLE_HS_LOG(DEBUG,"rawhdr =");
+            for (int i = 0; i < 25; i++) {
+                BLE_HS_LOG(DEBUG,"%02x ",evdata[i]);
+            }
+            BLE_HS_LOG(DEBUG,"\n");
+

+            break;
     default:
         BLE_HS_LOG(DEBUG, "\tUnknown LE event\n");
         break;

mwilliford, 2018-04-24:
[hci_debug_tests.txt](https://github.com/apache/mynewt-nimble/files/1940790/hci_debug_tests.txt)

mwilliford, 2018-05-11:
Tested with a lot less printing to the console, moved to RTT.  No luck with the 500byte tx->rx.  Here is what I see on Rx side:

06209 Extended adv: incompleted rssi=-42 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006209  length_data=226 data=00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df e0 e1
006210 Extended adv: incompleted rssi=-42 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006211  length_data=16 data=e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef f0 f1
006274 Extended adv: incompleted rssi=-45 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006274  length_data=226 data=00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df e0 e1
006275 Extended adv: incompleted rssi=-45 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006275  length_data=16 data=e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef f0 f1
006318 Extended adv: incompleted rssi=-42 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006318  length_data=226 data=00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df e0 e1
006319 Extended adv: incompleted rssi=-42 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006319  length_data=16 data=e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef f0 f1
006361 Extended adv: incompleted rssi=-42 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006362  length_data=226 data=00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df e0 e1
006363 Extended adv: incompleted rssi=-42 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006363  length_data=16 data=e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef f0 f1
006373 Extended adv: incompleted rssi=-43 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006373  length_data=226 data=00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df e0 e1
006374 Extended adv: incompleted rssi=-43 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006374  length_data=16 data=e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef f0 f1
006405 Extended adv: incompleted rssi=-43 txpower=127, pphy=1, sphy=1, sid=0, addr_type=1 addr=fe:bf:6c:7d:2c:59
006405  length_data=226 data=00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d [... cut ...] 1), not the 500 I attempted to transmit in this test case.  

I used my test repo above (I'll commit the latest so you can see it).  I have one side setup as tx, the other rx.  

Other notes:
At scale, i still get asserts and other problems with the device sees a lot of traffic, this is going to be a problem for my planned use case.  Stability is far off from where I need it to be.  My hope is that stability issues are only with ext_adv, as I mostly plan to use legacy for most traffic.  

Marcus

mwilliford, 2018-05-11:
@andrzej-kaczmarek 

With BTshell <--> BTshell, and a 900byte attempt, we see almost 100% of the attempts resulting in truncated transmissions.  Similar to what our test program struggles with:  https://github.com/mwilliford/mynewt-extadv-test

Here is the btshell output:

044449 Extended adv: truncated rssi=-33 txpower=127, pphy=1, sphy=1, sid=0, addr_type=0 addr=01:02:03:04:05:06
044452  length_data=226 data=0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05:0x06:0x07:0x08:0x09:0x00:0x01:0x02:0x03:0x04:0x05

andrzej-kaczmarek, 2018-05-11:
@mwilliford
We checked with your app and indeed we receive only data from 1st AUX from each chain in host. We do not know yet what causes this issue since with exactly the same NimBLE code we have no issues with scanning when using `btshell` (although we had to modify it so it does not print complete contents of AUX, as I described previously). We'll continue looking into this.

You also mentioned that you get various asserts - can you share which asserts you hit? Backtraces will be useful as well.

andrzej-kaczmarek, 2018-05-11:
@mwilliford 
A quick update: I compared radio behaviour between `btshell` and your app.
When using `btshell` scheduler wakes up precisely to receive each chain part. 
When using your app there's some weird behaviour of scheduler which wakes up just after 1st AUX_ADV_IND (this is likely NOT to receive chain since it's too early) and then again after 1st AUX_CHAIN_IND (this is likely to receive it, but it's too late).

We have no idea yet what causes this weird behaviour, will need to investigate it more later.

mwilliford, 2018-05-11:
@andrzej-kaczmarek 
ok, a few thoughts.  I should try to verify that my tasks are lower priority than bt_ll, however I think they should be lower pri already.

I had an idea about using GPIO signals to track on a scope when the bt_ll scheduler is in tx vs rx, any other thoughts on how this timing can be observed.

mwilliford, 2018-05-13:
@andrzej-kaczmarek 

Update, using bt_shell, it is possible to receive the 500 bytes transmitted (my test code on tx side only), ONLY if using RTT with btshell.  So, even bt_shell cannot receive the 500 bytes without RTT configured (my observation).

So, this further supports your observation above, which is that scheduler does not wake up at the right time when running my rx side code.  My code is now configured for RTT, I will continue to remove code on my example code until it matches BT_SHELL behavior.

mwilliford, 2018-05-13:
@andrzej-kaczmarek 
Fixed for 500 bytes!  It turns out that OS_MEMPOOL_POISON causes lots of timing issues, as one might expect.  My test program has this enabled, and I think this gets me working up to 500 bytes.  I'll now increase and continue to test.

Pushed updates to my test project, which correctly displays 500bytes on rx side now.

mwilliford, 2018-05-13:
Closing this ticket, as it is long and has a few issues in it, the core of which is solved.

Thanks @andrzej-kaczmarek 

I will try to open more focused bugs for any remaining issues that I hit regarding ext adv.

closed 2018-05-13 by mwilliford