apache/mynewt-nimble #46: ble_ll_scan.c calls ble_ll_scan_init twice, which corrupts g_os_mempool_list

mwilliford, 2018-04-16:
While troubleshooting why mpool cli command does not show anything beyond ble_ll_aux_scan_pool.

repro:

To watch it corrupt g_os_mempool_list, I did something like this following, where 0x20003648-0x2000364a is the pointer to the first of my missing mpool entries (via cli).  

So, I didn't know what was altering the memory, and I didn't own JTAG w/ TRACE, so I wrote a small MWU IRQ to catch whatever was altering this memory.  I then moved stack pointer back, so my Clion could show me the real offending stack.   I setup the MWU after all usual inits of mpools ran (at which point g_os_mpool_list was still accurate), then it hit (corrupted) when the ble stack was coming up.

Let me know if you have some more clever way to watch memory on the ARM for changes, I am just learning how to debug more advanced stuff.

So, ble_ll_scan_init triggered it, but on the 2nd runthrough.  So, my mpool was in the linked-list, but the 'new' mpool for ble_ll_scan_init was sorta replaced, and broke the linked list (cutoff any mpools after it).

This only occurs when BLE_LL_CFG_FEAT_LL_EXT_ADV is defined.

I will post a PR for the fix, verified it already.

------------- example code to catch this program in action------
    void MWU_IRQHandler( void );
    // start memory watch of 0x20003648 for debug
    NRF_MWU->REGION[2].START = (uint32_t) 0x20003648;
    NRF_MWU->REGION[2].END   = (uint32_t) 0x2000364a;
    //NRF_MWU->INTENCLR    = MWU_INTENCLR_REGION3WA_Clear << MWU_INTENSET_REGION2WA_Pos;
    NRF_MWU->INTENSET    = MWU_INTENSET_REGION2WA_Enabled << MWU_INTENSET_REGION2WA_Pos;
    NRF_MWU->REGIONENSET = MWU_REGIONENSET_RGN2WA_Enabled << MWU_REGIONENSET_RGN2WA_Pos;
    NVIC_SetVector(MWU_IRQn, (uint32_t)MWU_IRQHandler);
    NVIC_SetPriority(MWU_IRQn, (1 << __NVIC_PRIO_BITS) - 1);
    NVIC_ClearPendingIRQ(MWU_IRQn);
    NVIC_EnableIRQ(MWU_IRQn);

    /* As the last thing, process events from default event queue. */
    while (1) {
        os_eventq_run(os_eventq_dflt_get());
    }
}

void HardFault_Handler() {
    while (1) {
        if (hal_debugger_connected()) {
            /*
             * If debugger is attached, breakpoint here.
             */
            asm("bkpt");
        }
        NVIC_SystemReset();
    }
}

void MWU_IRQHandler( void )
{
    asm volatile(
    " tst lr,#4       \n"
            " ite eq          \n"
            " mrseq r0,msp    \n"
            " mrsne r0,psp    \n"
            " mov sp, r0       \n"
            " bkpt #1          \n"
    );
}

sjanc, 2018-04-16:
I've pushed PR with proposed os_mempool_clear functionality that should allow to fix this correctly 

https://github.com/apache/mynewt-core/pull/1030

sjanc, 2018-04-17:
All patches are now merged, please test if issue is fixed.

mwilliford, 2018-04-18:
It seems like this issue is fixed, thanks!

closed 2018-04-18 by mwilliford