apache/mynewt-nimble #103: BLE_EXT_ADV does not allow MESH to send messages

adityaxavier, 2018-05-19:
BLE_EXT_ADV does not allow bt_mesh_model_send to send the message.

The device is still able to receive the messages however.

This problem does not allow BLE and MESH to co-exist.

rymanluk, 2018-05-25:
Found the issue. #116
We used incorrect ADV PDU type when using BLE_EXT_ADV. Instead of ADV_NONCONN_IND we had ADV_SCAN_IND for ADV bearer. This cause mesh lib to drop the packet, which is rather correct.

adityaxavier, 2018-05-28:
Can you please test the patch with the sample Application I sent ? It seems with this patch MESH is not working. If there are any requirements on my end to change the Application, please let me know.

rymanluk, 2018-05-29:
In order this is helpful also for others please let me explain how I'm testing it with btshell. 

Goal:
1) configure two nodes so Device A can Generic On/Off server on Device B
2) configure BLE advertising instance on Device A so it can be connected by other peer devices.

Here is the target configuration:

    app=@apache-mynewt-core/apps/btshell
    bsp=@apache-mynewt-core/hw/bsp/nrf52840pdk
    build_profile=optimized
    syscfg=BLE_EXT_ADV=1:BLE_EXT_ADV_MAX_SIZE=1650:BLE_HCI_TRANSPORT_NIMBLE_BUILTIN=1:BLE_MESH=1:BLE_MESH_DEBUG=1:BLE_MESH_DEBUG_ACCESS=1:BLE_MESH_DEBUG_MODEL=1:BLE_MESH_DEBUG_NET=1:BLE_MESH_DEV_UUID=((uint8_t[16]){0x11, 0x33, 0}):BLE_MESH_SHELL=1:BLE_MESH_SHELL_MODELS=1:BLE_MONITOR_RTT=1:BLE_MULTI_ADV_INSTANCES=5:BLE_SM_LEGACY=1:BLE_SM_SC=1:LOG_LEVEL=0:MSYS_1_BLOCK_COUNT=48:RTT_BUFFER_SIZE_DOWN=64:SHELL_TASK=1

Note: I'm using this configuration for Device A and Device B. Make sure to use different BLE_MESH_DEV_UUID for those devices

Once built and flashed please do:

Device A shell:
1) select mesh
2) init
3) provision 0 1
4) app-key-add 0 0 1
5) mod-app-bind 1 0  0x1001  

Device B shell:
1) select mesh
2) init
3) provision 0 2

Go back to Device A shell:
6)  dst 2   // set destination address to 2
7)  app-key-add 0 0 1
8)  mod-app-bind 2 0  0x1000
9)  gen-onoff-set 5  // check console on Device B to see light state change.

// Now configure BLE advertising instance which will be used for BLE connection
10) select btshell
11) advertise-configure connectable=1 scannable=1 legacy=1 own_addr_type=random
12) advertise-set-adv-data flags=6 name=nimble
13) advertise-start

Now you can connect to this device from e.g. phone. In the same time you can control Device B over Mesh.

Hope that helps.

rymanluk, 2018-05-29:
@sjanc the above could be a part of the regression testing for releases. What do you think?

michal-narajowski, 2018-05-29:
@rymanluk good idea!

adityaxavier, 2018-05-30:
@rymanluk , Sorry it was a mistake on my end. The patch works !

closed 2018-05-30 by adityaxavier