apache/mynewt-nimble #166: [portability] NimBLE should be C99 compliant

turon, 2018-08-23:
Nimble currently can't be compiled with `-std=c99` flag passed to the compiler.

A number of errors will trigger when ISO C99 requirements are enforced.

andrzej-kaczmarek, 2018-08-23:
What configuration you use? Can you include some of the errors you see? I've just built host+controller with `-std=c99` without errors...

turon, 2018-08-23:
I'm using gcc 5.4.0 with `-Werror -std=c99 -pedantic-errors`.  I'm seeing warnings as errors, ISO C99 issues, and pedantic errors in nimble preventing integration with these compiler flags.

```
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Here are some example errors:

```
gcc -DHAVE_CONFIG_H -I. -I../../../include  -fPIC -I../../../include -I../../../examples/platforms -I../../../src/core -I../../../src/lib -D_GNU_SOURCE   -I../../../third_party/mynewt-nimble/repo/porting/npl/linux/include -I../../../third_party/mynewt-nimble/repo/porting/npl/linux/src -I../../../third_party/mynewt-nimble/repo/porting/nimble/include -I../../../third_party/mynewt-nimble/repo/nimble/include -I../../../third_party/mynewt-nimble/repo/nimble/host/include -I../../../third_party/mynewt-nimble/repo/nimble/host/src -I../../../third_party/mynewt-nimble/repo/nimble/host/services/ans/include -I../../../third_party/mynewt-nimble/repo/nimble/host/services/gap/include -I../../../third_party/mynewt-nimble/repo/nimble/host/services/gatt/include  -I/home/mturon/src_local/nest/openthread/ot-nimble/third_party/mbedtls -I/home/mturon/src_local/nest/openthread/ot-nimble/third_party/mbedtls/repo/include -I/home/mturon/src_local/nest/openthread/ot-nimble/third_party/mbedtls/repo/include/mbedtls -DMBEDTLS_CONFIG_FILE=\"mbedtls-config.h\"   -g -O2 -Wall -Wextra -Wshadow -Werror -std=c99 -pedantic-errors -D_BSD_SOURCE=1 -D_DEFAULT_SOURCE=1 -MT libopenthread_posix_a-ble_nimble.o -MD -MP -MF .deps/libopenthread_posix_a-ble_nimble.Tpo -c -o libopenthread_posix_a-ble_nimble.o `test -f 'ble_nimble.c' || echo './'`ble_nimble.c
In file included from ../../../third_party/mynewt-nimble/repo/porting/nimble/include/os/os.h:53:0,
                 from ../../../third_party/mynewt-nimble/repo/nimble/include/nimble/ble.h:26,
                 from ble_nimble.c:37:
../../../third_party/mynewt-nimble/repo/porting/nimble/include/os/os_mbuf.h:107:13: error: ISO C forbids zero-size array ‘om_databuf’ [-Wpedantic]
     uint8_t om_databuf[0];
             ^
In file included from ../../../third_party/mynewt-nimble/repo/porting/nimble/include/os/os.h:54:0,
                 from ../../../third_party/mynewt-nimble/repo/nimble/include/nimble/ble.h:26,
                 from ble_nimble.c:37:
../../../third_party/mynewt-nimble/repo/porting/nimble/include/os/os_mempool.h:71:29: error: ISO C99 doesn’t support unnamed structs/unions [-Wpedantic]
     SLIST_HEAD(,os_memblock);
                             ^
In file included from ble_nimble.c:37:0:
../../../third_party/mynewt-nimble/repo/nimble/include/nimble/ble.h:105:6: error: ISO C99 doesn’t support unnamed structs/unions [-Wpedantic]
     };
      ^
In file included from ../../../third_party/mynewt-nimble/repo/nimble/host/include/host/ble_hs.h:30:0,
                 from ble_nimble.c:38:
../../../third_party/mynewt-nimble/repo/nimble/include/nimble/hci_common.h:910:46: error: ISO C forbids zero-size array ‘params’ [-Wpedantic]
     struct hci_le_subev_direct_adv_rpt_param params[0];
                                              ^
In file included from ../../../third_party/mynewt-nimble/repo/nimble/host/include/host/ble_hs_adv.h:24:0,
                 from ../../../third_party/mynewt-nimble/repo/nimble/host/include/host/ble_gap.h:32,
                 from ../../../third_party/mynewt-nimble/repo/nimble/host/include/host/ble_hs.h:33,
                 from ble_nimble.c:38:
../../../third_party/mynewt-nimble/repo/nimble/host/include/host/ble_uuid.h:95:33: error: ISO C does not permit named variadic macros [-Wvariadic-macros]
 #define BLE_UUID128_INIT(uuid128...)    \
                                 ^
../../../third_party/mynewt-nimble/repo/nimble/host/include/host/ble_uuid.h:107:36: error: ISO C does not permit named variadic macros [-Wvariadic-macros]
 #define BLE_UUID128_DECLARE(uuid128...) \
                                    ^
```

ccollins476ad, 2018-08-23:
Anonymous structs and unions were added in C11.  I agree that conforming to an older specification improves portability, but C11 is already seven years old! :)

turon, 2018-08-23:
@ccollins476ad I can look into pushing the project I'm integrating into up to `-std=c11`, but what are your thoughts on being able to pass `-Werror` and `-pedantic-errors`?

ccollins476ad, 2018-08-23:
I think achieving ISO C compliance will be quite a challenge.  The nimble codebase knowingly uses quite a few gcc/clang-isms (e.g., `__attribute__((packed))`).

Standards conformance is certainly a good thing, but it was not viewed as a requirement when nimble was being developed.  So, in my opinion, it is not worth the effort required at this point.  Of course, it all depends on whether and why someone needs nimble to be standards compliant.

Citrullin, 2018-12-07:
@ccollins476ad I think it is always better to be C99 compliant. I already had a lot of pthread issues which were solved by strict C99 compliant code and compiler usage. I heard the same by other experiences embedded software developer. They all are developing in strict C99 standard.

Avamander, 2020-10-30:
There's no point in regressing to strict C99 when we have C11.