apache/mynewt-nimble #188: Change a logging macro definition to avoid macro argument prescan

nutka7, 2018-09-18:
I have a suggestion regarding logging macros definitions from `nimble/host/include/host/ble_hs_log.h`.

Currently there are defined as follows:
```
#define BLE_HS_LOG(lvl, ...) \
    MODLOG_ ## lvl(LOG_MODULE_NIMBLE_HOST, __VA_ARGS__)

#define BLE_HS_LOG_ADDR(lvl, addr)                      \
    BLE_HS_LOG(lvl, "%02x:%02x:%02x:%02x:%02x:%02x",    \
               (addr)[5], (addr)[4], (addr)[3],         \
               (addr)[2], (addr)[1], (addr)[0])
```

The problem I encountered when porting mynewt-nimble to a new platform is that `lvl` argument of the `BLE_HS_LOG_ADDR` is prescanned. Very often external code uses the `DEBUG` macro (e.g. setting it to 1 or 0) and as a result nimble fails to compile.

Do you think it would be beneficial to change the `BLE_HS_LOG_ADDR` definition to avoid `lvl` argument expansion? For example, although a bit more verbose, it could be defined as  

```
#define BLE_HS_LOG_ADDR(lvl, addr)                      \
    MODLOG_ ## lvl(LOG_MODULE_NIMBLE_HOST,              \
                   "%02x:%02x:%02x:%02x:%02x:%02x",     \
                   (addr)[5], (addr)[4], (addr)[3],     \
                   (addr)[2], (addr)[1], (addr)[0])
```

ccollins476ad, 2018-09-18:
Thanks, @nutka7.  I see what you mean.  That is a bummer...

Thanks for suggesting the fix; it looks good to me.  I will submit a PR shortly.

ccollins476ad, 2019-05-24:
This was fixed by #388.

closed 2019-05-24 by ccollins476ad