apache/mynewt-nimble #191: bletiny application source is missing apps/ repo.

gladborn, 2018-09-20:


andrzej-kaczmarek, 2018-09-20:
`bletiny` was deprecated some time ago and recently removed from repository - use `btshell` instead which supports all the features available in NimBLE

gladborn, 2018-09-21:
How to compile btshell ? I can't find any Makefile.

andrzej-kaczmarek, 2018-09-21:
the same way as any other app for Mynewt - you need to create target and build it using `newt` tool
if you did not use Mynewt before, start here: http://mynewt.apache.org/latest/

gladborn, 2018-09-21:
Do you have any test application for Linux to test ported example code of linux ?

mynewt-nimble-master\porting\examples\linux

andrzej-kaczmarek, 2018-09-21:
on Linux you can use BlueZ - there are tools like `bluetoothctl` there or just use UI

gladborn, 2018-09-21:
Thanks. Helpful

closed 2018-09-21 by gladborn