apache/mynewt-nimble #217: Add missing GATT descriptor UUIDs

microbuilder, 2018-10-10:
The current release of nimble includes a define for the [client characteristic configuration descriptor (0x2902)](https://github.com/apache/mynewt-nimble/blob/master/nimble/host/include/host/ble_gatt.h#L46). None of the other [official descriptor UUIDs](https://www.bluetooth.com/specifications/gatt/descriptors) are present in `ble_gatt.h`. 

In order to make use of the [characteristic user description (0x2901)](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_user_description.xml), I currently have to define this myself or hard-code the value, both of which seem unfortunate.

Would it be appropriate to add at least the more commonly used ones, or does the current code operate under the premise that defines are only added as they are needed in the library itself or the larger mynewt-core repository? As limiting as that may be, I understand the goal of avoiding code bloat.

andrzej-kaczmarek, 2018-10-11:
I think this is good idea, but I do not have strong opinion on where we should put this.
Perhaps separate include like `ble_gatt_uuid.h` with definitions for all defined GATT 16-bit UUIDs? In Zephyr all UUIDs are defined in `uuid.h` but I do not like it since this is the place where generic UUIDs handling is defined.

So perhaps that is not very helpful, but if there is PR with some proposal I can check it.

microbuilder, 2018-10-13:
`ble_gatt_uuid.h` makes sense to me since it's obvious where to find the values and keep up to date. I can put a PR together if you can give me an example for descriptors and characteristic/service define names to follow?