apache/mynewt-nimble #254: BLE Central: Requests with same opcode in quick succession do not work

dhrishi, 2018-11-30:
If in a BLE GATT central, two or more requests with same opcode are sent (for e.g. Write or Notify), the requests go through and responses are received. Although, the responses for all the requests except for the first one are not sent above to the app (or upper layers)

This is because, currently ble_gattc_extract_first_by_conn_op()  extracts all the matching procs and then they are removed from the list. Ideally, it should extract the first matching proc (as its name suggests).

dhrishi, 2018-11-30:
Modified ble_gattc_extract_by_conn_op() which is used from ble_gattc_extract_first_by_conn_op() to accept max_procs and extract the appropriate number of procs, as required.

Submitted a pull request for this https://github.com/apache/mynewt-nimble/pull/253

rymanluk, 2018-12-07:
#253 is merged, we can close it. Thanks!

closed 2018-12-07 by rymanluk