apache/mynewt-nimble #263: GATT: Analyze and fix failing testcases when using AutoPTS

michal-narajowski, 2018-12-11:
This is a general issue that groups all issues with GATT testcases when using AutoPTS. AutoPTS for Nimble is still work in process.

Test Case | Result |  
-- | -- | --
GATT/CL/GAR/BI-05-C | BTP ERROR | Read Characteristic Value - Insufficient Encryption Key Size
GATT/CL/GAR/BI-22-C | BTP ERROR | Read Multiple Characteristic Values - Insufficient Encryption Key Size
GATT/CL/GAW/BI-06-C | BTP ERROR | Write Characteristic Value - Insufficient Encryption Key Size
GATT/CL/GAW/BI-13-C | BTP ERROR | Write Long Characteristic Value - Insufficient Encryption Key Size
GATT/SR/GAD/BV-03-C | BTP ERROR | Find Included Services - from Server
GATT/SR/GAI/BV-01-C | BTP ERROR | Characteristic Value Indication - by Server
GATT/SR/GAN/BV-01-C | BTP ERROR | Characteristic Value Notification - by Server
GATT/SR/GAR/BI-05-C | BTP ERROR | Read Characteristic Value - Insufficient Encryption Key Size
GATT/SR/GAR/BI-11-C | BTP ERROR | Read Using Characteristic UUID - Insufficient Encryption Key Size
GATT/SR/GAR/BI-17-C | BTP ERROR | Read Long Characteristic Value - Insufficient Encryption Key Size
GATT/SR/GAR/BI-22-C | BTP ERROR | Read Multiple Characteristic Values - Insufficient Encryption Key Size
GATT/SR/GAR/BV-06-C | BTP ERROR | Read Characteristic Descriptors - from Server
GATT/SR/GAT/BV-01-C | BTP ERROR | Handle Value Indication - Client Timeout
GATT/SR/GAW/BI-06-C | BTP ERROR | Write Characteristic Value - Insufficient Encryption Key Size
GATT/SR/GAW/BI-13-C | BTP ERROR | Write Long Characteristic Value - Insufficient Encryption Key Size
GATT/CL/GAI/BV-01-C | BTP TIMEOUT | Confirm Characteristic Value Indication - by Client
GATT/CL/GAN/BV-01-C | BTP TIMEOUT | Characteristic Value Notification - to Client
GATT/CL/GAR/BI-14-C | BTP TIMEOUT | Read Long Characteristic Value - Invalid Handle
GATT/CL/GAR/BI-17-C | BTP TIMEOUT | Read Long Characteristic Value - Insufficient Encryption Key Size
GATT/CL/GAR/BI-18-C | BTP TIMEOUT | Read Multiple Characteristic Values - Read Not Permitted
GATT/CL/GAW/BI-07-C | BTP TIMEOUT | Write Long Characteristic Value - Invalid Handle
GATT/CL/GAW/BV-09-C | BTP TIMEOUT | Write Long Characteristic Descriptors - by Client
GATT/SR/GAR/BI-06-C | FAIL | Read Characteristic by UUID - Read Not Permitted Response
GATT/SR/GAR/BI-10-C | FAIL | Read Using Characteristic UUID - Insufficient Authentication
GATT/SR/GAD/BV-01-C | INCONC | Discover All Primary Services - from Server
GATT/SR/GAD/BV-02-C | INCONC | Discover Primary Services by Service UUID - from Server
GATT/SR/GAR/BV-04-C | INCONC | Read Long Characteristic Value - from Server
GATT/SR/GAR/BV-05-C | INCONC | Read Multiple Characteristic Values - from Server
GATT/SR/GAS/BV-01-C | INCONC | Service Changed Characteristic - from Server
GATT/SR/GAW/BV-01-C | INCONC | Write Without Response - to Server
GATT/CL/GAR/BV-04-C | PTS TIMEOUT | Read Long Characteristic Value - by Client
GATT/CL/GAW/BI-09-C | PTS TIMEOUT | Write Long Characteristic Value - Invalid Offset
GATT/SR/GAC/BV-01-C | PTS TIMEOUT | Server Configuration - of Server

michal-narajowski, 2019-01-07:
Update 07.01.2019:

Test Case | Result |  
-- | -- | --
GATT/CL/GAI/BV-01-C | BTP TIMEOUT | Confirm Characteristic Value Indication - by Client
GATT/CL/GAN/BV-01-C | BTP TIMEOUT | Characteristic Value Notification - to Client

michal-narajowski, 2019-02-07:
Update 07.02.2019:

Test Case | Result |  
-- | -- | --
GATT/CL/GAT/BV-02-C | BTP ERROR | Write Characteristic Value - Server Timeout
GATT/CL/GAT/BV-01-C | FAIL | Read Characteristic Value - Server Timeout
GATT/CL/GAI/BV-01-C | PTS TIMEOUT | Confirm Characteristic Value Indication - by Client
GATT/CL/GAN/BV-01-C | PTS TIMEOUT | Characteristic Value Notification - to Client

michal-narajowski, 2019-02-12:
Update 12.02.2019:

Test Case | Result |  
-- | -- | --
GATT/CL/GAI/BV-01-C | BTP TIMEOUT | Confirm Characteristic Value Indication - by Client
GATT/CL/GAN/BV-01-C | BTP TIMEOUT | Characteristic Value Notification - to Client

rymanluk, 2019-10-23:
Ia it still valid?

michal-narajowski, 2021-05-27:
This is an old issue, nowadays we have rare failures so we can keep an issue per failed test.

closed 2021-05-27 by michal-narajowski