apache/mynewt-nimble #265: L2CAP: Analyze and fix failing testcases when using AutoPTS

michal-narajowski, 2018-12-11:
This is a general issue that groups all issues with L2CAP testcases when using AutoPTS. AutoPTS for Nimble is still work in process.

Test Case | Result |  
-- | -- | --
L2CAP/LE/CPU/BV-01-C | BTP TIMEOUT | Verify that the IUT is able to send the connection parameter update Request to tester when acting as a slave device.
L2CAP/LE/CPU/BI-01-C | PTS TIMEOUT | Verify that the IUT is able to reject a request for connection parameter update with illegal parameters.

michal-narajowski, 2019-01-07:
Update 07.01.2019

Test Case | Result |  
-- | -- | --
L2CAP/LE/CPU/BV-01-C | BTP TIMEOUT | Verify that the IUT is able to send the connection parameter update Request to tester when acting as a slave device.

michal-narajowski, 2019-02-07:
Update 07.02.2019:

Test Case | Result |  
-- | -- | --
L2CAP/COS/CFC/BV-03-C | INCONC | Verify that the IUT can correctly reassemble data received from the Lower Tester which is greater than the IUT LE-frame size.
L2CAP/COS/CFC/BV-04-C | INCONC | Verify that the IUT can receive unsegmented data correctly.

michal-narajowski, 2019-02-12:
Update 12.02.2019:

All tests pass

rymanluk, 2019-10-23:
We can close it then

closed 2019-10-23 by rymanluk