apache/mynewt-nimble #436: Extended Advertising with 2M secondary PHY

br101, 2019-05-13:
I am trying to receive Extended Advertisements from a Nordic nRF52 using the Nordic SDK (ble_app_rscs) with another Nordic nRF52 device with Nimble (using btshell and blehci). It works, when the Nordic SDK advertiser sends with 1M on both the primary and the secondary PHY, but when it has BLE_GAP_PHY_2MBPS on the secondary, the advertisements are not seen on the Nimble side...

Is 2M not implemented yet for extended advertisements? Do I need to activate a configuration option? Or am I missing something?

andrzej-kaczmarek, 2019-05-13:
2M and Coded are disabled by default in NimBLE controller, you can enable them by setting `BLE_LL_CFG_FEAT_LE_2M_PHY: 1` or `BLE_LL_CFG_FEAT_LE_CODED_PHY: 1` respectively

br101, 2019-05-13:
yes, that works! great! :-)

closed 2019-05-13 by br101