apache/mynewt-nimble #446: [controller] hardfault exception in ble_ll_scan_aux_data_unref

lance-proxy, 2019-05-24:
Exception occurs on following line:

 https://github.com/apache/mynewt-nimble/blob/a586a56b4923b057ccff3f2f3900a7828db57e16//nimble/controller/src/ble_ll_scan.c#L1303

Suspect the `aux_data` is pointing at garbage / out of range address, which when dereferenced leads to a memory access error and hardfault.

This has been very hard to repro on the bench, it has occurred very occasionally in a test environment. Test units are scanning, advertising and connecting. `BLE_EXT_ADV` is enabled. Also have the follow Link Layer sysconfigs disabled:

```
   BLE_LL_EXT_ADV_AUX_PTR_CNT: 0
  BLE_LL_CFG_FEAT_LE_2M_PHY: 0
  BLE_LL_CFG_FEAT_LE_CODED_PHY: 0
  BLE_LL_CFG_FEAT_LE_CSA2: 0
  BLE_LL_CFG_FEAT_LE_ENCRYPTION: 0
  BLE_LL_CFG_FEAT_LE_PING: 0
  BLE_LL_CFG_FEAT_LL_PRIVACY: 0
```

rymanluk, 2019-05-28:
@lance-proxy Interesting thing here is that BLE_LL_EXT_ADV_AUX_PTR_CNT is 0 in your test environment.  In such a case aux_data shall be always NULL.  I'm not sure now if we are looking into good place ...

Anyway, I have just tested such configuration and it works for me even I started extended scanner.

Couple of questions:
1. Could you send your target configuration ?
2. Did you manage to reproduce it and then connect debugger to the device?
3. Do you start extended scanner as well?

Note, using extended scanning and BLE_LL_EXT_ADV_AUX_PTR_CNT =0 does not really makes sense, but for unit tests this is actually good :)

btw: I pushed https://github.com/apache/mynewt-nimble/pull/449 but this is not solving anything, it makes code cleaner.

lance-proxy, 2019-05-28:
@rymanluk, I have been update to catch this in the debugger so far, but before you posted this update, I actually went and removed setting `BLE_LL_EXT_ADV_AUX_PTR_CNT` to 0 and since then none of the devices in our test environment have hit this exception. Extended scanning is being started (but only just have support for `legacy_event_type`). Will keep an eye out for this and keep you posted on further updates.

lance-proxy, 2019-05-29:
@rymanluk spoke too soon :(. I'm still seeing this issued with the`BLE_LL_EXT_ADV_AUX_PTR_CNT` setting of zero above removed. I've attached a file that contains all our targets BLE configs:
[ble_config.txt](https://github.com/apache/mynewt-nimble/files/3230255/ble_config.txt)

In the meantime, I will continue trying to see if I can catch this in the debugger and provide further updates.

lance-proxy, 2019-06-03:
@rymanluk, as discussed here is some pieces of GDB debug output after trapping this exception in the debugger for my target:

Backtrace:
```
(gdb) bt
#0  hal_watchdog_tickle () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_watchdog.c:119
#1  0x00011f64 in os_default_irq (tf=0x2000ff10) at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:267
#2  0x00013d5e in os_default_irq_asm () at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/m4/HAL_CM4.s:260
#3  <signal handler called>
#4  ble_ll_scan_aux_data_unref (aux_data=0x5003f0f0) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_scan.c:1303
#5  0x0002ac36 in ble_ll_init_rx_isr_end (rxbuf=rxbuf@entry=0x2000d07c <g_ble_phy_rx_buf+4> "D#z\b\223\006\037\310\b\tpr-5c21\023\377\004\003", crcok=<optimized out>, ble_hdr=0x2000f040 <g_ble_phy_data+24>)
    at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_conn.c:2995
#6  0x00027c5a in ble_ll_rx_end (rxbuf=rxbuf@entry=0x2000d07c <g_ble_phy_rx_buf+4> "D#z\b\223\006\037\310\b\tpr-5c21\023\377\004\003", rxhdr=rxhdr@entry=0x2000f040 <g_ble_phy_data+24>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:1016
#7  0x0003063e in ble_phy_rx_end_isr () at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1056
#8  ble_phy_isr () at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1237
#9  <signal handler called>
#10 0x00000000 in ?? ()
#11 0x20000228 in _acUpBuffer ()
```
Global variable inspections:
```
p g_ble_phy_data
$1 = {
  phy_stats_initialized = 1 '\001', 
  phy_txpwr_dbm = 0 '\000', 
  phy_chan = 37 '%', 
  phy_state = 1 '\001', 
  phy_transition = 0 '\000', 
  phy_transition_late = 0 '\000', 
  phy_rx_started = 1 '\001', 
  phy_encrypted = 0 '\000', 
  phy_privacy = 0 '\000', 
  phy_tx_pyld_len = 35 '#', 
  phy_txtorx_phy_mode = 1 '\001', 
  phy_cur_phy_mode = 1 '\001', 
  phy_bcc_offset = 0 '\000', 
  rx_pwr_compensation = 0 '\000', 
  phy_aar_scratch = 0, 
  phy_access_address = 2391391958, 
  rxhdr = {
    {
      rxinfo = {
        flags = 131, 
        channel = 37 '%', 
        handle = 0 '\000', 
        rssi = -78 '\262', 
        phy = 1 '\001', 
        phy_mode = 1 '\001', 
        user_data = 0x5003f0f0
      }, 
      txinfo = {
        flags = 131 '\203', 
        offset = 0 '\000', 
        pyld_len = 37 '%', 
        hdr_byte = 0 '\000'
      }
    }, 
    beg_cputime = 119339527, 
    rem_usecs = 2
  }, 
  txend_arg = 0x0, 
  txend_cb = 0x0, 
  phy_start_cputime = 119339518
}
```
p g_ble_ll_data
$3 = {
  ll_supp_features = 4110, 
  ll_state = 3 '\003', 
  ll_num_acl_pkts = 4 '\004', 
[... 553 lines cut ...]
      }
    }
  }
}
```
Memory pools:
```
p ext_adv_pool
$5 = {
  mp_block_size = 64, 
  mp_num_blocks = 5, 
  mp_num_free = 5, 
  mp_min_free = 5, 
  mp_flags = 0 '\000', 
  mp_membuf_addr = 536923716, 
  mp_list = {
    stqe_next = 0x2000cbf0 <sensor_notify_evt_pool>
  }, 
  {
    slh_first = 0x2000ce44 <ext_adv_mem>
  }, 
  name = 0x3c7b5 "ble_ll_aux_scan_pool"
}

Please let me know what I else I should dump (I will keep the DUT powered on spinning in an infinite loop in os_default_isr). Please let me know the best way to dump`ble_ll_stats` in GDB.

rymanluk, 2019-06-04:
@lance-proxy  looks like you are actually trying to connect. Scan responses in this case has bit different path.  Could you print `g_ble_ll_conn_create_sm` as well? to print 'ble_ll_stats' you should use 'p' as you are doing with others globals. 

So it looks like something is writing to `scansm ->cur_aux_data`, but I have no idea what yet. If it would be possible to reproduce with a debugger we could set BLE_LL_EXT_ADV_AUX_PTR_CNT =0 and set a write watchpoint on  `scansm ->cur_aux_data` to find out what is it.

Keep looking into it....

simonratner, 2019-06-04:
A quick search turned up something suspicious: https://github.com/apache/mynewt-nimble/blob/master/nimble/controller/src/ble_ll_conn.c#L560

Is it meant to be `unref`ed twice? This change was introduced in https://github.com/apache/mynewt-nimble/commit/2df0d01fe354ef289d4eb2857e8dbe6b5a1859f4.

rymanluk, 2019-06-04:
`ble_ll_scan_aux_data_unref()` returns number of refs and second `unref` happens only when ref is >0. Actually I should put inside that `if ()` an `assert(scansm->cur_aux_data->ref_cnt == 1)`, because this is the only case.

rymanluk, 2019-06-04:
@lance-proxy could it be that connection cancel has been done in your test scenario? If so, maybe that patch will help: https://github.com/apache/mynewt-nimble/pull/460 , as there might be possible race.

lance-proxy, 2019-06-04:
@rymanluk it is possible that the either the client or server can async cancel the connection in our scenario.

Here is `g_ble_ll_conn_create_sm` as requested:
```
(gdb) p g_ble_ll_conn_create_sm
$1 = (struct ble_ll_conn_sm *) 0x0
```
Do I need to enable the LL stats as I'm not seeing it populated with any data fields?
```
(gdb) p ble_ll_stats
$2 = {<No data fields>}
(gdb) 
```
Also, how would watchpoint monitoring of `scansm ->cur_aux_data` work given it its a local variable?

lance-proxy, 2019-06-05:
@rymanluk update:  after implementing your patch above still getting the exception with `aux_data=0x5003f0f0 ` so this unfortunately does not resolve the source of the corruption of `aux_data`.

rymanluk, 2019-06-05:
@lance-proxy regarding watch point on `scansm ->cur_aux_data` - note that `scansm` is a pointer to global `g_ble_ll_scan_sm`

Regarding  magic `aux_data=0x5003f0f0`  - it is interesting. Could you reproduce it with BLE_LL_EXT_ADV_AUX_PTR_CNT =0  and see if it is the same there?

rymanluk, 2019-06-05:
@lance-proxy Do you have asserts enabled? could you also try with this fix: https://github.com/apache/mynewt-nimble/pull/463

lance-proxy, 2019-06-05:
@rymanluk, yes same `aux_data=0x5003f0f0` when I previously had `BLE_LL_EXT_ADV_AUX_PTR_CNT =0 `.  I don't have `BLE_LL_VND_EVENT_ON_ASSERT` enabled but if was getting into the code path you added a `return` for in #463, it looks like it would still assert according to https://github.com/apache/mynewt-nimble/blob/54fc157c665cbe1a40311ca1602d09c324ada1c0/nimble/controller/include/controller/ble_ll.h#L57 - and we do have assertions (i.e `assert(0)`) working generally.

rymanluk, 2019-06-06:
@lance-proxy  ok, good to know, but that means, we need watch point to see what is writing it.

lance-proxy, 2019-06-07:
@rymanluk still unsure on how to set watchpoint on `scansm ->cur_aux_data` (and also given the crash is random so I can't predict when it will happen so any pointers on how to do that would be good.

Also, looking forward to testing your potential fix in https://github.com/apache/mynewt-nimble/pull/466 once its merged into NimBle master.

rymanluk, 2019-06-07:
@lance-proxy the thing is that when `BLE_LL_EXT_ADV_AUX_PTR_CNT =0` then `g_ble_ll_scan_sm.cur_aux_data` shall be always `NULL` as we shall never , so we need set conditional watchpoint on that.

When it comes to #466, I believe it will not gonna help as problem occurs when controller does not schedule any aux_ptr. Anyway, when that patch is ready to test I will notify you and will take of [WIP] from the PR name.

lance-proxy, 2019-06-07:
@rymanluk I updated to latest NimBLE `master`, set BLE_LL_EXT_ADV_AUX_PTR_CNT =0 & then set watchpoint on `g_ble_ll_scan_sm.cur_aux_data`.

However, the same exception occurred(`aux_data=0x5003f0f0`) *without*  the watchpoint getting triggered .See GDB trace below:
```
(gdb) watch g_ble_ll_scan_sm.cur_aux_data
Hardware watchpoint 1: g_ble_ll_scan_sm.cur_aux_data
(gdb) continue
Continuing.
(gdb) bt
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
hal_watchdog_tickle () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_watchdog.c:118
118	    NRF_WDT->RR[0] = WDT_RR_RR_Reload;
(gdb) bt
#0  hal_watchdog_tickle () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_watchdog.c:118
#1  0x000121f4 in os_default_irq (tf=0x2000ff10) at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:268
#2  0x00013fee in os_default_irq_asm () at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/m4/HAL_CM4.s:260
#3  <signal handler called>
#4  ble_ll_scan_aux_data_unref (aux_data=0x5003f0f0)
```
Is there something wrong I'm doing, or could it be that `g_ble_ll_scan_sm` (or scansm ->cur_aux_data) is getting corrupted via another mechanism?

rymanluk, 2019-06-08:
I think I know what is happening.  I should actually notice it when you said that `g_ble_ll_conn_create_sm` is NULL. I think there is a problem as we already have no connection (`connsm` is NULL) and `connsm->scansm` is already trash in `ble_ll_init_rx_isr_end`

I believe that https://github.com/apache/mynewt-nimble/pull/469 should fix it.

lance-proxy, 2019-06-10:
@rymanluk that's good news you have identified the underlining issue! I will test your patch asap , look forward to seeing #469 merged soon.

lance-proxy, 2019-06-12:
@rymanluk good news since updating to NimBle top of.trunk (with #469) none of our test units have exhibited this exception. So I've closed this issue, thanks again for looking into this and for the fix.

closed 2019-06-12 by lance-proxy