apache/mynewt-nimble #532: [controller] Hardfault exception in LL

lance-proxy, 2019-07-20:
Debug unit is running current top of NimBLE master: https://github.com/apache/mynewt-nimble/commit/8b9cde1df640b64e1f4aa1d5d376b781c152f8ae

See the GDB stack trace below (for details of our NimBLE configuration refer to #446)):
```
(gdb) bt
#0  hal_system_reset () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_system.c:56
#1  0x000121bc in os_default_irq (tf=0x2000ff30) at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:275
#2  0x00013f6e in os_default_irq_asm () at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/m4/HAL_CM4.s:260
#3  <signal handler called>
#4  os_memblock_get (mp=0x2000a284 <os_msys_1_mempool>) at repos/apache-mynewt-core/kernel/os/src/os_mempool.c:346
#5  0x00012adc in os_mbuf_get (omp=0x2000a278 <os_msys_1_mbuf_pool>, leadingspace=leadingspace@entry=0) at repos/apache-mynewt-core/kernel/os/src/os_mbuf.c:135
#6  0x00012b0c in os_mbuf_get_pkthdr (omp=<optimized out>, user_pkthdr_len=<optimized out>) at repos/apache-mynewt-core/kernel/os/src/os_mbuf.c:169
#7  0x00013380 in os_msys_get_pkthdr (dsize=dsize@entry=14, user_hdr_len=user_hdr_len@entry=20) at repos/apache-mynewt-core/kernel/os/src/os_msys.c:163
#8  0x000281ac in ble_ll_rxpdu_alloc (len=len@entry=14) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:335
#9  0x00028490 in ble_ll_rx_end (rxbuf=rxbuf@entry=0x2000d1e8 <g_ble_phy_rx_buf+4> "\003\fS\327\346N@X\241\060\033\326sD\001\004\003\256*\003\001\004\003/", rxhdr=rxhdr@entry=0x2000f1a8 <g_ble_phy_data+24>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:1057
#10 0x000310b2 in ble_phy_rx_end_isr () at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1070
#11 ble_phy_isr () at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1251
#12 <signal handler called>
#13 0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```
Note: everything has been stable with the same app build/configuration pinned to this prvious commit in NimBLE:
https://github.com/apache/mynewt-nimble/tree/b6da0700d25a75124793d2ed9079610260befd51

I suspect there is some race condition occurring someone in the LL.

lance-proxy, 2019-07-20:
Dumping some LL variables here to aid debug this crash:
```
(gdb) p g_ble_ll_data
$3 = {
  ll_supp_features = 4110, 
  ll_state = 4 '\004', 
  ll_num_acl_pkts = 4 '\004', 
  ll_acl_pkt_size = 255, 
  ll_pref_tx_phys = 0 '\000', 
  ll_pref_rx_phys = 0 '\000', 
  ll_rfclk_state = 2 '\002', 
  ll_xtal_ticks = 49, 
  ll_rfclk_start_time = 1362672, 
  ll_rfclk_timer = {
    bsp_timer = 0x2000d940 <nrf52_hal_timer5>, 
    cb_func = 0x30925 <ble_ll_xcvr_rfclk_timer_exp>, 
    cb_arg = 0x0, 
    expiry = 1412703, 
    link = {
      tqe_next = 0x2000f12c <g_ble_ll_sched_timer>, 
      tqe_prev = 0x0
    }
  }, 
  ll_evq = {
    evq = {
      evq_owner = 0x2000dc5c <g_ble_ll_task>, 
      evq_task = 0x2000dc5c <g_ble_ll_task>, 
      evq_list = {
        stqh_first = 0x0, 
        stqh_last = 0x2000dce0 <g_ble_ll_data+52>
      }
    }
  }, 
  ll_wfr_timer = {
    bsp_timer = 0x0, 
    cb_func = 0x0, 
    cb_arg = 0x0, 
    expiry = 0, 
    link = {
      tqe_next = 0x0, 
      tqe_prev = 0x0
    }
  }, 
  ll_rx_pkt_ev = {
    ev = {
      ev_queued = 0 '\000', 
      ev_cb = 0x28111 <ble_ll_event_rx_pkt>, 
      ev_arg = 0x0, 
      ev_next = {
        stqe_next = 0x0
      }
    }
  }, 
  ll_rx_pkt_q = {
    stqh_first = 0x0, 
    stqh_last = 0x2000dd10 <g_ble_ll_data+100>
  }, 
  ll_tx_pkt_ev = {
    ev = {
      ev_queued = 0 '\000', 
      ev_cb = 0x280a9 <ble_ll_event_tx_pkt>, 
      ev_arg = 0x0, 
      ev_next = {
        stqe_next = 0x0
      }
    }
  }, 
  ll_tx_pkt_q = {
    stqh_first = 0x0, 
    stqh_last = 0x2000dd28 <g_ble_ll_data+124>
[... 161 lines cut ...]
  cur_aux_data = 0x0, 
  phy_data = {{
      phy = 1 '\001', 
      own_addr_type = 1 '\001', 
      scan_filt_policy = 0 '\000', 
      configured = 1 '\001', 
      scan_type = 1 '\001', 
      scan_chan = 37 '%', 
      scan_itvl = 48, 
      scan_window = 48, 
      scan_win_start_time = 1412804, 
      next_event_start = 0
    }, {
      phy = 3 '\003', 
      own_addr_type = 0 '\000', 
      scan_filt_policy = 0 '\000', 
      configured = 0 '\000', 
      scan_type = 0 '\000', 
      scan_chan = 0 '\000', 
      scan_itvl = 0, 
      scan_window = 0, 
      scan_win_start_time = 0, 
      next_event_start = 0
    }}
}

andrzej-kaczmarek, 2019-07-20:
seems like mpool is broken since this happens in os code on alloc, not sure what could cause this yet.
is it hard to reproduce this issues? there are quite a lot of changes between last stable revision and current master so will need to look through them, but would it be perhaps possible for you to find commit which breaks it using `git bisect`? that would help a lot.

rymanluk, 2019-07-22:
@lance-proxy bisecting this is the first prio but then you could also look at the os_msys_1_mempool when it happens. There is `compiler/gdbmacros/mbuf.gdb` script which could help (command `mn_msys1_print`)
Just make sure you have this patch (https://github.com/apache/mynewt-core/pull/1922)

lance-proxy, 2019-07-22:
@andrzej-kaczmarek yes the issue occurs quite frequently especially in an environment with lots of  connections recycling quickly.

@rymanluk with OS_MEMPOOL_POISON enabled, I get the following assertion:

```
#0  __assert_func (file=file@entry=0x0, line=line@entry=0, func=func@entry=0x0, e=e@entry=0x0) at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:176
#1  0x00010680 in os_mempool_poison_check (start=start@entry=0xf5ae <__assert_func+30>, mp=<optimized out>) at repos/apache-mynewt-core/kernel/os/src/os_mempool.c:78
#2  0x00010766 in os_memblock_get (mp=0x10681 <os_mempool_poison_check+60>) at repos/apache-mynewt-core/kernel/os/src/os_mempool.c:357
#3  0x0000ff6e in os_mbuf_get (omp=0x200096f8 <os_msys_1_mbuf_pool>, leadingspace=<optimized out>) at repos/apache-mynewt-core/kernel/os/src/os_mbuf.c:135
#4  0x0001089c in os_msys_get (dsize=dsize@entry=83, leadingspace=leadingspace@entry=0) at repos/apache-mynewt-core/kernel/os/src/os_msys.c:135
#5  0x0001f410 in ble_ll_rxpdu_alloc (len=len@entry=195) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:367
#6  0x0001f6c8 in ble_ll_rx_end (
    rxbuf=rxbuf@entry=0x2000c310 <g_ble_phy_rx_buf+4> "\003\301\244\214\230\366sL\255\357֬\353\335\236|\220i\355\026Gu>G\361[\")\233}j}Ǎ\366\067\354h\n\373#y\340=I\350\022Z\376\205K\211\004\tK\323\r\266\033M\265\301\332\f/\220\065\022\333\327`\274'\273\032\217\335\307:\204\205f\270>ˀRۅG\255-\177\255\a\001\035 \260\275\331\025\310\006_\225\363\206\206 \376L\227/\263\200ځ\353\214\301ü\363ޥ[\177\205\261\235\230e", <incomplete sequence \325>, rxhdr=rxhdr@entry=0x2000ea6c <g_ble_phy_data+24>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:1057
#7  0x000282ea in ble_phy_rx_end_isr () at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1070
#8  ble_phy_isr () at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1251
```

with OS_MEMPOOL_GUARD & OS_MEMPOOL_CHECK enabled I get a strange hang.

Next, I am going to find the offending commit with `git bisect` & bring the new gdb macro.

lance-proxy, 2019-07-22:
@rymanluk I brought in current MyNewt master with your `mn_mysys1_print` fix. The attached file contains the output from `mn_mysy1_print` when hitting this crash:

[mn_msys1_print.txt](https://github.com/apache/mynewt-nimble/files/3418744/mn_msys1_print.txt)

lance-proxy, 2019-07-23:
@rymanluk @andrzej-kaczmarek here is the result of my `git bisect`:

```
git bisect log
git bisect start
# bad: [8b9cde1df640b64e1f4aa1d5d376b781c152f8ae] nimble/mesh: Fix null pointer access on prov_init
git bisect bad 8b9cde1df640b64e1f4aa1d5d376b781c152f8ae
# good: [b6da0700d25a75124793d2ed9079610260befd51] nimble/ll: Fix cancel connection creation
git bisect good b6da0700d25a75124793d2ed9079610260befd51
# good: [d9992ac5a888fb4ba93feb4d9d51e988d3b31256] nimble/host: Simplify lookup for LE Meta event handler
git bisect good d9992ac5a888fb4ba93feb4d9d51e988d3b31256
# bad: [0e9b96d21763c0d9e874a20f90815b19595e96c1] nimble/ll: Add assert to detect not completed chained advertising report
git bisect bad 0e9b96d21763c0d9e874a20f90815b19595e96c1
# bad: [d512d353a021605b901740e3bd3281c547a9a9c2] nimble/ll: Add static checks for mbuf data alignment
git bisect bad d512d353a021605b901740e3bd3281c547a9a9c2
# good: [0da4decf660bd857dcea1d8cfd75a9a92dc2012b] nimble/ll: Rename functions for consistency
git bisect good 0da4decf660bd857dcea1d8cfd75a9a92dc2012b
# good: [68527f9f239967b9e54b344d4d3fa7a48f7a4253] nimble/ll: Cleanup ble_ll_rxpdu_alloc
git bisect good 68527f9f239967b9e54b344d4d3fa7a48f7a4253
# bad: [99df12df07e193964a980dc74c8c79b15f6f0a89] nimble/phy/nrf51: Optimize PDU copy
git bisect bad 99df12df07e193964a980dc74c8c79b15f6f0a89
# bad: [f19002e909b108585b2a1747e7cf45b81b1f85fe] nimble/phy/nrf52: Optimize PDU copy
git bisect bad f19002e909b108585b2a1747e7cf45b81b1f85fe
# first bad commit: [f19002e909b108585b2a1747e7cf45b81b1f85fe] nimble/phy/nrf52: Optimize PDU copy
```

andrzej-kaczmarek, 2019-07-23:
@lance-proxy thanks, I see the issue in that commit now. could you please test with this provisional fix: https://github.com/apache/mynewt-nimble/pull/534

lance-proxy, 2019-07-23:
@andrzej-kaczmarek I tested with your patch. Unfortunately, still seems to hang/crash though its slightly different:

```
(gdb) bt
#0  hal_system_reset () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_system.c:56
#1  0x0000f698 in os_default_irq (tf=0x2000ff20) at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:274
#2  0x0001134a in os_default_irq_asm () at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/m4/HAL_CM4.s:260
#3  <signal handler called>
#4  0x00011c36 in memcpy (dst=0x2000c719 <g_ble_phy_tx_buf+9>, src=0x208aca19, n=23, n@entry=27) at repos/apache-mynewt-core/libc/baselibc/src/memcpy.c:31
#5  0x00010120 in os_mbuf_copydata (m=0x20007658 <os_msys_1_data+96>, off=<optimized out>, off@entry=0, len=27, dst=<optimized out>) at repos/apache-mynewt-core/kernel/os/src/os_mbuf.c:463
#6  0x000208fa in ble_ll_adv_legacy_pdu_make (hdr_byte=0x2000ff97 "B\204", <incomplete sequence \324>, pducb_arg=0x2000d484 <g_ble_ll_adv_sm+252>, dptr=0x2000c719 <g_ble_phy_tx_buf+9> "\b\tpr-422f\023\377\004\003") at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_adv.c:473
#7  ble_ll_adv_pdu_make (dptr=0x2000c713 <g_ble_phy_tx_buf+3> "\034", <incomplete sequence \373>, pducb_arg=0x2000d484 <g_ble_ll_adv_sm+252>, hdr_byte=0x2000ff97 "B\204", <incomplete sequence \324>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_adv.c:515
#8  0x00028438 in ble_phy_tx (pducb=pducb@entry=0x20879 <ble_ll_adv_pdu_make>, pducb_arg=pducb_arg@entry=0x2000d484 <g_ble_ll_adv_sm+252>, end_trans=<optimized out>) at repos/apache-mynewt-nimble/nimble/drivers/nrf52/src/ble_phy.c:1724
#9  0x00020056 in ble_ll_adv_tx_start_cb (sch=0x2000d4e8 <g_ble_ll_adv_sm+352>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_adv.c:1061
#10 0x00026e18 in ble_ll_sched_execute_item (sch=0x2000d4e8 <g_ble_ll_adv_sm+352>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_sched.c:1501
#11 ble_ll_sched_run (arg=<optimized out>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_sched.c:1536
#12 0x0000ee52 in hal_timer_chk_queue (bsptimer=bsptimer@entry=0x2000d0a4 <nrf52_hal_timer5>) at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_timer.c:277
#13 0x0000eea2 in hal_rtc_timer_irq_handler (bsptimer=0x2000d0a4 <nrf52_hal_timer5>) at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_timer.c:387
#14 nrf52_timer5_irq_handler () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_timer.c:440
```

andrzej-kaczmarek, 2019-07-23:
@lance-proxy sorry, I made a stupid mistake there. I pushed updated and tested fix, please try once more, I am pretty sure this will fix the issue.

michal-narajowski, 2019-07-23:
I was able to reproduce this and @andrzej-kaczmarek's patch fixed this issue for me.

lance-proxy, 2019-07-23:
@andrzej-kaczmarek just tested your latest patch fix, and yes it looks like it has resolved the issue. No crashes so far based on my testing. So I think the PR is good to merge 👍

andrzej-kaczmarek, 2019-07-23:
@lance-proxy thank you, I've just merged it. we'll include it in upcoming 1.2 release of nimble.

closed 2019-07-24 by lance-proxy