apache/mynewt-nimble #541: Assertion in ble_phy_isr

lance-proxy, 2019-07-24:
Following fix for #532, based on our testing I am now seeing an assertion occurring here:
https://github.com/apache/mynewt-nimble/blob/master/nimble/drivers/nrf52/src/ble_phy.c#L1246

Stack trace:
```
024349 Assert @ 0x27a77
024349 [ts=190226548us, mod=203 level=1] event_recorder: event_record_append() 30 byte record
024349 Unhandled interrupt (3), exception sp 0x2000bcc0
024349  r0:0x00000001  r1:0xffffffff  r2:0x00000037  r3:0x00000001
024349  r4:0x2000cdcc  r5:0x00002000  r6:0x2000bbec  r7:0x00000001
024349  r8:0xffffffff  r9:0x00000035 r10:0x0001eecd r11:0x00000000
024349 r12:0x00000000  lr:0x00010e57  pc:0x0001079e psr:0x61000000
024349 ICSR:0x00421803 HFSR:0x40000000 CFSR:0x01000000
024349 BFAR:0xe000ed38 MMFAR:0xe000ed34

(gdb) bt
#0  hal_system_reset () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_system.c:56
#1  0x0000f35c in os_default_irq (tf=0x2000ffc8) at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:276
#2  0x00010f92 in os_default_irq_asm () at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/m4/HAL_CM4.s:260
#3  <signal handler called>
#4  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

lance-proxy, 2019-07-24:
@andrzej-kaczmarek @rymanluk, let me know what else I can do help debug this one. Hope we can fix this for the 1.2 NimBLE release.

rymanluk, 2019-07-26:
@lance-proxy Could you test with https://github.com/apache/mynewt-nimble/pull/544/commits/f747c6e447abf27a1e286b693ed75e4468690334

lance-proxy, 2019-07-26:
@rymanluk so far with your patch things look good, no observed asserts.

closed 2019-07-29 by lance-proxy