apache/mynewt-nimble #551: Not able to resolve RPA public address

prasad-alatkar, 2019-08-05:
While advertising I am trying to set 'own_address_type' to `BLE_OWN_ADDR_RPA_PUBLIC_DEFAULT`, however peer device is not able to connect with the device, dhkey check is failing(error as `BLE_SM_ERR_DHKEY`) and peer device address is shown as all 0s (not able to resolve peer identity address ?).

    I am using **NimBLE host only** implementation, IO_CAP is **`keyboard only`**, `sm_our_key_dist/sm_their_key_dist  = BLE_SM_PAIR_KEY_DIST_ID`.

    Even with the API `ble_hs_id_infer_auto`, if I try to set `privacy = 1`, I am facing the same issue.

rymanluk, 2019-10-23:
Can you provide btsnoop logs?

prasad-alatkar, 2019-10-24:
Hi @rymanluk I am attaching device side logs as well as scanner side logs here. 
[ESP32_rpa_addr_log.txt](https://github.com/apache/mynewt-nimble/files/3767788/ESP32_rpa_addr_log.txt)
[Scanner_Log 2019-10-24 18-55-48.txt](https://github.com/apache/mynewt-nimble/files/3767789/Scanner_Log.2019-10-24.18-55-48.txt)

prasad-alatkar, 2019-11-28:
Hi @rymanluk  , Can you please let me know if NimBLE host supports "Host-based privacy" (Vol 3, part C, section 10.7.1.2) ? I understand "controller-based privacy" is supported. Just wanted to build my understanding related to resolving RPA in NimBLE host.

marcogmaia, 2020-05-27:
Have you solved this issue? I can't find a way to resolve host based RPA either

prasad-alatkar, 2020-05-28:
Hi @marcogmaia AFAIK, If you are using 'NimBLE controller' then there should not be any issue with resolving RPA. We have ported only "NimBLE Host" on ESP32 where we were facing issues with RPA so we have tried our hands with `Host based RPA`, repo: https://github.com/espressif/esp-nimble

marcogmaia, 2020-05-28:
Thank you, I found instructions there, but I think I've found a bug around c++ includes in ble_hs_pvcy.h.
Gonna open an issue there, and a PR.

closed 2020-07-13 by prasad-alatkar