apache/mynewt-nimble #552: ble_gattc_notify_custom cuts off message

HendrikVE, 2019-08-05:
I'm trying to send terminal output via the notification mechanism (`ble_gattc_notify_custom`). The problem is that only the first 20 characters are sent, everything else is getting cut off and never gets sent. Do I have to divide the message into packets myself or does a more suitable function exist for my use case?

mfarag13, 2019-08-05:
Are you increasing the MTU?
Default value is 23, 3 are ATT header and the remaining 20 are data.
If the data is being sent is greater than MTU size, it truncates to MTU size, hence sending 20 byte only.

HendrikVE, 2019-08-05:
No, I'm not touching the MTU. Is there no built-in mechanism in nimble to let nimble send multiple packets if the message does not fit in to the MTU?

mfarag13, 2019-08-05:
I don't think so, you have to develop your own mechanism to fragment packets on MTU. Or You can exchange the mtu size.
ble_att_set_preferred_mtu ---> to increase your default MTU.
ble_gattc_exchange_mtu ---> to exchange MTU with peer device.

HendrikVE, 2019-08-12:
Thanks, I was able to work around this problem with a change of the MTU, but afterwards I got another problem. I was sending a few messages in a really short time interval and ble_gattc_notify_custom failed to deliver most of my messages. I had to slow down my sending thread with roughly 50 ms for each sending operation. What could be a cleaner solution for this? Aren't messages queued internally?

verjus, 2020-10-04:
I am posting here so as to not open a duplicate issue. 

I am having the exact same issue, which I solved in a similar manner. I maintain a list of packets representing the entire payload that I need to transfer. The total payload is greater than the MTU but not each packet in the list. I send the first packet via:

```
    struct os_mbuf *om = ble_hs_mbuf_from_flat(node->packet.data, node->packet.dataSize);
    int rc = ble_gattc_notify_custom(conn_handle, alexa_notify_handle, om);

```
and wait for the  BLE_GAP_EVENT_NOTIFY_TX event. When I receive this event,
I check event->notify_tx.status and send the next packet in the packet list. If there are still packets left in the list, it will trigger again a notify_tx even until there are no packets left.

This appears to work. However, when I stress test it by sending 338 bytes with an MTU of 33, it fails after the 12th packet. The return code is notify_tx.status = 6, which seems to correspond to BLE_HS_ENOMEM. This would indicate a heap exhaustion situation. I don't understand how this could be possible. Aren't the buffers returned to the heap after each call to ble_gattc_notify_custom? So what am I doing wrong?

Here is how it looks like:
This is the list of packets I am trying to send with a series of notifications (no indicate):

Packet [1/13] contains: 60 00 00 01 52 18 0a cf  02 0a 24 0a 0f 41 6c 65 
                        78 61 2e 44 69 73 63 6f  76 65 72 79 12 11 
Packet [2/13] contains: 60 14 1b 44 69 73 63 6f  76 65 72 2e 52 65 73 70 
                        6f 6e 73 65 12 a6 02 0a  a3 02 0a 0e 41 31 
Packet [3/13] contains: 60 24 1b 4f 52 32 44 55  50 37 4a 30 35 39 5a 12 
                        0a 4e 69 78 69 65 54 69  6d 65 72 1a 0b 4e 
Packet [4/13] contains: 60 34 1b 69 78 69 65 20  54 69 6d 65 72 22 0e 41 
                        58 53 20 49 6e 64 75 73  74 72 69 65 73 5a 
Packet [5/13] contains: 60 44 1b 1d 0a 0e 41 6c  65 78 61 49 6e 74 65 72 
                        66 61 63 65 12 06 41 6c  65 72 74 73 1a 03 
Packet [6/13] contains: 60 54 1b 31 2e 31 5a 55  0a 0e 41 6c 65 78 61 49 
                        6e 74 65 72 66 61 63 65  12 1a 41 6c 65 78 
Packet [7/13] contains: 60 64 1b 61 2e 47 61 64  67 65 74 2e 53 74 61 74 
                        65 4c 69 73 74 65 6e 65  72 1a 03 31 2e 30 
Packet [8/13] contains: 60 74 1b 22 22 0a 08 0a  06 74 69 6d 65 72 73 0a 
                        0a 0a 08 74 69 6d 65 69  6e 66 6f 0a 0a 0a 
Packet [9/13] contains: 60 84 1b 08 77 61 6b 65  69 6e 66 6f 62 72 0a 01 
                        30 12 40 34 32 37 38 44  43 34 35 34 33 31 
Packet [10/13] contains: 60 94 1b 41 33 32 38 41  37 46 41 32 30 44 38 35 
                         36 36 31 36 46 34 44 32  42 37 34 41 44 33 
Packet [11/13] contains: 60 a4 1b 34 36 46 35 32  37 42 43 33 34 33 35 36 
                         43 33 37 32 31 33 32 31  38 44 37 43 43 1a 
Packet [12/13] contains: 60 b4 1b 01 31 22 0e 41  31 4f 52 32 44 55 50 37 
                         4a 30 35 39 5a 2a 0a 4e  69 78 69 65 54 69 
Packet [13/13] contains: 60 c8 11 6d 65 72 32 0c  34 34 30 30 34 39 30 46 
                         36 41 31 30 

after the 11th packet, I get:

preparing to send chunk of data size= 30
sending packet number 12 <<<<<<<<
60 b4 1b 01 31 22 0e 41  31 4f 52 32 44 55 50 37 
  4a 30 35 39 5a 2a 0a 4e  69 78 69 65 54 69 
Sending notification now
GATT procedure initiated: notify; att_handle=12
event type= 0x000d
notification update; status=0   <<<<<<<<<< SUCCESS
Notify tx connection_handle = 64
Notify tx attr handle = 12
Notify tx indication  = 0
PacketList_send called with:
firstPacket = 0
Resp code = 0

But the following packet (which is the last by design) fails:
Preparing to send chunk of data size= 20
sending packet number 13  <<<<<<<<
60 c8 11 6d 65 72 32 0c  34 34 30 30 34 39 30 46 
  36 41 31 30 
Sending notification now
GATT procedure initiated: notify; att_handle=12
event type= 0x000d
notification update; status=6 <<<<<<<<<< FAILURE
Notify tx connection_handle = 64
Notify tx attr handle = 12
Notify tx indication  = 0
PacketList_send called with:
firstPacket = 0
Resp code = 6 <<<
Retry

and never succeed after that.

ccollins476ad, 2020-10-04:
> This would indicate a heap exhaustion situation. I don't understand how this could be possible. Aren't the buffers returned to the heap after each call to ble_gattc_notify_custom?

The buffers are not immediately freed after the call.  Assuming you are using a combined host-controller, there are roughly two steps each time your application sends a notification:
1. Application passes data to the host.
2. Controller sends data to the destination device.

`ble_gattc_notify_custom()` implements the first step.  This step is very fast.  Essentially, the host just hands the controller task a pointer to the buffer to send.  The second task is much slower because the controller must schedule its transmits according to how BLE is specified and how the connection is configured.

It sounds like the packets are sitting in the controller's transmit queue.  Unfortunately, I'm not sure there's a perfect solution to this problem.  Some possibly-good-enough solutions are: 1) increase the buffer count, or 2) set a timer that continues the transmission on expiry.

verjus, 2020-10-04:
Thank you for taking the time to answer my question. This is very helpful. I thought there was something wrong with my fragmentation code.

For now, I will increase the value of MSYS_1_BLOCK_COUNT as you suggested, which is indeed set by default to (12). This explains why I am failing at packet 13. 

Meanwhile, I read in [this](https://github.com/espressif/esp-idf/issues/4328) post that os_msys_num_free() can be used to monitor the number of available free blocks. The same post suggests monitoring this value before sending the next notification.

I may try to change my code accordingly to determine when the number of free blocks has increased before sending the next batch of notifications. This adds a lot of complexity but would ensure that the code keeps working with small MTUs and relatively large buffers.

I wonder if this logic should be incorporated as a feature in NimBLE, i.e. automatically and transparently fragment and send a long notification based on the negotiated MTU.

konteris, 2023-12-07:
Was this feature added or we still have to manually fragment notification data into mtu size packets and send it?