apache/mynewt-nimble #556: att_svr memory corruption: entry_mem is freed without relocating existing entries

haukepetersen, 2019-08-15:
# How to reproduce
I initialized the GATT server in the following sequence:
- init some included services, e.g.: `ble_svc_gap_init()`, `ble_svc_gatt_init()`
- start the GATT server: `ble_gatts_start()`
- add some custom services: 'ble_gatts_count_cfg()`, `ble_gatts_add_svcs()`
- (re)start the GATT server, to load additional services: 'ble_gatts_start()`

This will lead to hard-faults on my platform (`nrf52dk`, RIOT, nimble controller).
Used this RIOT branch: https://github.com/haukepetersen/RIOT/tree/opt_nimble_gattsinitpr
NimBLE version: same on master and on 997dad8c9fc549e64b6c54eafcec109d92789418

# Problem
When initializing the GATT server, and thereby implicitly the ATT server, the ATT server frees and (re-)allocates the internal memory it uses for storing the servers entries (`ble_att_svr_entry_mem`). The problem is, that if `ble_gatts_start()` is called multiple times, the ATT server frees `ble_att_svr_entry_mem` which still contains valid entries. This freed memory is eventually re-used by other components in the system, leading to corrupted entries in the ATT server, and eventually making a system hard-fault...

Simple test: add `assert((void *)entry >= (void *)ble_att_svr_entry_mem);` into `ble_att_svr_find_by_uuid()` -> this assertion is triggered when executing the sequence sketched above

I am not so sure about a viable fix though: simply removing the `ble_att_svr_free_start_mem()` call from `ble_att_svr_start()` will fix this issue, but it sure enough opens a memory leak :-)

I guess we probably must probably do something like this in `ble_att_svr_start()`:
- allocate the new entry memory
- move all existing entries from the old entry memory into the new one
- and only then free the old entry memory again?

andrewleech, 2019-09-09:
@haukepetersen Have your tried your suggested fix at all, or had any luck with any other workaround?

We've run into the same/similar problem while integrating nimble into micropython, originally assuming we'd be able to add extra services by registering them then calling `ble_gatts_start()`

The api docs (https://mynewt.apache.org/latest/network/docs/ble_hs/ble_gattc.html#c.ble_gatts_start) don't exactly make it clear whether appending new services to the existing set is supposed to be supported or not? I would certainly vote for it as a very desirable feature!

Either way having a memory corruption easily triggered by running `ble_gatts_start` without previously running `ble_gatts_reset` is undesirable.

haukepetersen, 2019-09-20:
No, I have not yet looked into this topic any further. It is on my list, but not at the top right now :-)