apache/mynewt-nimble #586: l2cap coc: Add timer guard on the receiver side

rymanluk, 2019-09-26:
In current implementation L2CAP CoC server returns credits when complete SDU is received.
Once SDU is received user gets notification and we required from the user to call `ble_l2cap_recv_ready()` with new buffer for data - this is the moment when we give but credits to the other side.

Problem here is when for some reason part of the SDU is lost. In such a case server stays in incorrect state.

To fix it we need some timer guard for incoming LE Frames which should be stopped when complete SDU is received. If we start missing LE Frames we, buffers should be flashed and already used credits shall be returned.

We might consider using ble_hs_timer

michal-narajowski, 2019-11-19:
I found a related issue when the client sends an SDU larger than we can process. In that case, we return an error but we never return credits so the client cannot send any more data and the server is left in an invalid state.

KKopyscinski, 2020-08-03:
Both cases should be fixed by this patch: https://github.com/apache/mynewt-nimble/pull/851

rymanluk, 2020-08-05:
@KKopyscinski 
thanks for the patch.
However, after more thinking I'm not sure we really want the guard timer on data on receiver side as this is not according to spec.
It is too bad I did not put in this issue logs. I recall we used stress tests to reproduce it  - could you please retest so we can look into this problem one more time?

What @michal-narajowski  wrote should be fixed, but as per specification:

"If the SDU length field value exceeds the receiver's MTU, the receiver shall
disconnect the channel"

KKopyscinski, 2020-08-05:
I periodically tested my solution with blestress - for larger amounts of data, frames were indeed missing, and triggered event was returning used credits.

I think at least second part, described by @michal-narajowski, is fine - we are disconnecting, but before we disconnect, we return credits. This part doesn't use timer.

rymanluk, 2020-08-05:
return tickets before disconnection is not needed since channel is not used anyway.

KKopyscinski, 2020-08-05:
Ok, so disconnecting channel and returning error is sufficient - as it was. About first part, I wonder if it's actually  possible to  lose frames - shouldn't that been taken care of by link layer? So packets should be retransmitted if they haven't arrieved.

rymanluk, 2020-08-05:
Exactly, LL should re transmit packets, so this is why I'm asking you to reproduce the issue as we need to understand what happend that server did not receive the packet.

We need btmon logs on both sides plus host logs to find it out.

rymanluk, 2020-08-05:
regarding second issue:
Indeed it was already solved:

commit 7f0a9c950388b5f5c08632b6e0017e8e408d6cc0
Author: Łukasz Rymanowski <lukasz.rymanowski@codecoup.pl>
Date:   Mon Jan 27 21:46:52 2020 +0100

    nimble/l2cap: Disconnect CoC when too big SDU is sent by peer
    
    This is to satisfy PTS and we actually planned to do so anyway.

sjanc, 2021-05-27:
is this stil valid?

KKopyscinski, 2021-06-09:
The problem with missing frames is solved: both host and controller are working fine. Blestress was just configured not well enough: setting `BLE_ACL_BUF_SIZE: MYNEWT_VAL_BLE_L2CAP_COC_MPS+4` solves the issue - test is not stuck, all data is transferred properly. Basically, we need to use ACL data packets that are able to contain MPS and HCI data header. For default BLE_L2CAP_COC_MPS size (MSYS_1_BLOCK_SIZE-8 = 286) default BLE_ACL_BUF_SIZE of 255 is just to small

closed 2021-06-09 by KKopyscinski