apache/mynewt-nimble #615: Nimble Host API for Linux

jonsmirl, 2019-10-18:
I'm working on porting micropython BLE support around to various microcontrollers. Most of these platforms use Nimble so that is pretty easy to do.  But micropython also works on Linux - both desktop and embedded. Using BLE support through DBUS makes everything a whole lot bigger.

So is these a way to get the Nimble Host API running on Linux? This would be a way clearer solution. It would also have a whole lot less memory consumption.  I guess Nimble would need a driver that calls the Linux kernel's HCI API? How would you sort out L2CAP being inside kernel?

Something like this would also appear to be useful for Newt OS development by letting you do partial development on your desktop.

turon, 2019-10-19:
NimBLE has already been ported to Linux via the NPL layer.  It works fairly
well, though there were some key fixes after the 1.2.0 release, so be sure
to evaluate on master.

See:
https://github.com/apache/mynewt-nimble/tree/master/porting/examples/linux
_____________________________
Martin Turon  |  Nest Labs

jonsmirl, 2019-10-19:
Thanks, that looks just like what I need.

Does it co-exist with Bluez?  Can you run Bluetooth audio simultaneously while using nimble for BLE?

jonsmirl, 2019-10-19:
Might want to add this to the readme...

sudo setcap 'cap_net_raw,cap_net_admin+eip' nimble-linux
That will let the app run without needing sudo.

Or you can add yourself to the bluetooth group.

sjanc, 2019-10-23:
Hi,

Two separate hosts are not be able to share single controller.
It can co-exist with BlueZ in sense that BlueZ is using one controller, and nimble is using another on same box.

closed 2021-05-27 by sjanc