apache/mynewt-nimble #631: 修改开发板BLE的服务的UUID，只能重启手机系统后才能重新更新到最新的？ 重启开发板和重启APP没用。

xumn3348, 2019-10-30:
手机蓝牙APP连接（或重连） 外围设备后， 发的是 attr req，而不是group type req， 这样如果动态修改了属性的信息，譬如uuid， 在app上是感知不到的， 需要重启手机后，APP才会发group type req请求更新这些信息。
不清楚nordic 的闭源协议栈如何处理这个问题，但nimBLE协议栈的确对这个问题不能健壮处理。

utzig, 2019-10-30:
Mandarin might be the language with most speakers in the world, but I don't think most people here understand it. Anyway google translate seems to do a good job: 

"After the mobile phone Bluetooth APP is connected (or reconnected) to the peripheral device, the attr req is sent instead of the group type req, so if the attribute information is dynamically modified, such as uuid, it is not perceived on the app, and the device needs to be restarted. The APP will send a group type req request to update this information.
It is not clear how nordic's closed source protocol stack handles this problem, but the nimBLE protocol stack does not handle this problem robustly."

xumn3348, 2019-11-01:
Thank you for the perfect translating。Pls focus on the problem, not the language。

rymanluk, 2019-11-04:
Hi @xumn3348, first of all thank you for reporting issue, also I would like to ask you to do it in English (also in other issues) so we don't have to focus on the language :)

Back to the issue:
We need more information as I'm not sure I do understand it correctly. It is also good to get related logs which shows the problem ( best would be to get btmon https://www.codecoup.pl/blog/support-for-btmon-in-mynewt/)

I guess the problem is that Mynewt is peripheral device in your use case. Also I guess that Mynewt changes its GATT Database when is disconnected from the bonded mobile phone, and after re-connection, mobile phone is not aware about GATT database changes- is that correct use case? 
 If not please provide the use case and the issue you are seeing.

xumn3348, 2019-11-04:
@rymanluk ,  i am Sorry to use Chinese。It is exactly the use case you just said 。 i am not sure if this is a bug for the peripheral device which uses Nimble stack。more importantly, how to update my IPhone without rebooting  when  the GATT DB changes?

sjanc, 2021-05-27:
please reopen with btmon logs attached if this is stil an issue

closed 2021-05-27 by sjanc