apache/mynewt-nimble #633: nimBLE 和手机pair不能正常进行？

xumn3348, 2019-10-30:
ble_hs_hci_acl_tx_now， 长度比较大的包调用ble_hs_tx_data包竟然发不到ble_phy_tx这边。怀疑是fragment机制有问题。

rymanluk, 2019-11-04:
translation:
"Ble_hs_hci_acl_tx_now, the packet with a relatively large length calls the ble_hs_tx_data package and does not reach the ble_phy_tx side. Suspected that there is a problem with the fragment mechanism."

@xumn3348  please provide logs showing the problem.

xumn3348, 2019-11-07:
thanks for your translation。 i use Apple Iphone 6S, and build secure connection(SC)， but after exchanging  pairing information,  the IPhone sends public key to slave board。 but slave board can't send  public key to the IPhone back 。 i checked the code and found that it can't enter ble_phy_rx_start_isr and ble_phy_rx_end_isr, i guess the IPhone's mechanism of SC can't deal with it。but it's just my guess。

In addition, legacy pairing mode is okay.

xumn3348, 2019-11-09:
i found the result is that Applie Iphone 6S 's  supervision timeout is 72， it 's the IPhone 6s  which disconnected with the slave board。 I changed the value from 72 to 255,  the SC works。

closed 2020-07-24 by rymanluk