apache/mynewt-nimble #643: blehci example, Hardware Error HCI event (and reset) when a long HCI command issued

bethjonesbluesky, 2019-11-10:
I'm seeing a Hardware Error (0x10) HCI event (then the board gets reset) in the `blehci` example when I send a long HCI command whose length is bigger than or equal to 61. Is there a configuration value I need to adjust for this? I'm using nrf52840.

It looks like `ble_hci_uart_state.rx_type` value is not valid, which caused the error (please see the gdb output at the end of this posting), but I'm not sure why the value is set wrong.

Note: I'm using bluez on host and had to comment out a few lines from Line 722 in ble_ll_hci.c because bluez calls a legacy advertisement HCI command when I register the controller:
<pre>
722         /*
723         if (hci_adv_mode == ADV_MODE_LEGACY) {`
724             return false;
725         }
726         */
</pre>

Here are some test snippets:
<pre>
$ newt target show
targets/blehci
    app=@apache-mynewt-nimble/apps/blehci
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=debug
    syscfg=BLE_EXT_ADV=1:BLE_EXT_ADV_MAX_SIZE=255:BLE_LL_CFG_FEAT_LE_2M_PHY=1:BLE_PUBLIC_DEV_ADDR=(uint8_t[6]){0x12, 0x34, 0x56, 0x78, 0x9a, 0xbc}:LOG_LEVEL=0
targets/my_blinky_sim
    app=apps/blinky
    bsp=@apache-mynewt-core/hw/bsp/native
    build_profile=debug
targets/nrf52840_boot
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
</pre>
<pre>
$ sudo hcitool -i hci0 cmd 08 036 00 00 00 00 01 00 00 40 00 01 00 00 66 55 44 33 22 11 00 08 01 00 01 01 00
< HCI Command: ogf 0x08, ocf 0x0036, plen 25
  00 00 00 00 01 00 00 40 00 01 00 00 66 55 44 33 22 11 00 08
  01 00 01 01 00
> HCI Event: 0x0e plen 5
  01 36 20 00 04
$ sudo hcitool -i hci0 cmd 08 037 00 03 00 38 \
> 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f \
> 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f \
> 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f \
> 30 31 32 33 34 35 36 37
< HCI Command: ogf 0x08, ocf 0x0037, plen 60
  00 03 00 38 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
  30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 30 31 32 33
  34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 30 31 32 33 34 35 36 37
> HCI Event: 0x0e plen 4
  01 37 20 00
$ sudo hcitool -i hci0 cmd 08 037 00 03 00 39 \
> 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f \
> 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f \
> 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f \
> 30 31 32 33 34 35 36 37 38
< HCI Command: ogf 0x08, ocf 0x0037, plen 61
  00 03 00 39 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
  30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 30 31 32 33
  34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 30 31 32 33 34 35 36 37
  38
> HCI Event: 0x10 plen 1
  01
</pre>

<pre>
< HCI Command: LE Set Extended Advertising Parameters (0x08|0x0036) plen 25
        Handle: 0x00
        Properties: 0x0000
        Min advertising interval: 160.000 msec (0x0100)
        Max advertising interval: 10240.000 msec (0x4000)
        Channel map: 37 (0x01)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: 11:22:33:44:55:66 (OUI 11-22-33)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
        TX power: 8 dbm (0x08)
        Primary PHY: LE 1M (0x01)
        Secondary max skip: 0x00
        Secondary PHY: LE 1M (0x01)
        SID: 0x01
        Scan request notifications: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 5
      LE Set Extended Advertising Parameters (0x08|0x0036) ncmd 1
        Status: Success (0x00)
        TX power (selected): 4 dbm (0x04)
@ RAW Close: hcitool
@ RAW Open: hcitool (privileged) version 2.22
@ RAW Close: hcitool
@ RAW Open: hcitool (privileged) version 2.22
@ RAW Close: hcitool
@ RAW Open: hcitool (privileged) version 2.22
< HCI Command: LE Set Extended Advertising Data (0x08|0x0037) plen 60
        Handle: 0x00
        Operation: Complete extended advertising data (0x03)
        Fragment preference: Fragment all (0x00)
        Data length: 0x38
        Unknown EIR field 0x31: 32333435363738393a3b3c3d3e3f303132333435363738393a3b3c3d3e3f303132333435363738393a3b3c3d3e3f30
        31 32 33 34 35 36 37                             1234567
> HCI Event: Command Complete (0x0e) plen 4
      LE Set Extended Advertising Data (0x08|0x0037) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool
@ RAW Open: hcitool (privileged) version 2.22
@ RAW Close: hcitool
@ RAW Open: hcitool (privileged) version 2.22
@ RAW Close: hcitool
@ RAW Open: hcitool (privileged) version 2.22
< HCI Command: LE Set Extended Advertising Data (0x08|0x0037) plen 61
        Handle: 0x00
        Operation: Complete extended advertising data (0x03)
        Fragment preference: Fragment all (0x00)
        Data length: 0x39
        Unknown EIR field 0x31: 32333435363738393a3b3c3d3e3f303132333435363738393a3b3c3d3e3f303132333435363738393a3b3c3d3e3f30
        31 32 33 34 35 36 37 38                          12345678
> HCI Event: Hardware Error (0x10) plen 1
        Code: 0x01
</pre>

<pre>
Breakpoint 1, ble_hci_uart_rx_pkt_type (data=85 'U') at repos/apache-mynewt-nimble/nimble/transport/uart/src/ble_hci_uart.c:416
416     {
(gdb) n
419         ble_hci_uart_state.rx_type = data;
(gdb)
421         switch (ble_hci_uart_state.rx_type) {
(gdb) p ble_hci_uart_state.rx_type
$1 = 85 'U'
(gdb) n
468             ble_hci_uart_sync_lost();
(gdb)
</pre>

sjanc, 2019-11-12:
Hi,

Are you using it via ACM? If so please disable MSD mode on your DK.
Do do it execute msddisable under JLinkExe.

bethjonesbluesky, 2019-11-14:
Thanks! it solved the problem.

closed 2019-11-14 by bethjonesbluesky