apache/mynewt-nimble #649: controller: data length extension for TX not applied

haukepetersen, 2019-11-14:
# Description
When sending data between 2 nodes using a L2CAP COC, I noticed that the controllers data length extension is not working as I expected it to: although the option is enabled for the controller and the maximum packet size (`BLE_LL_MAX_PKT_SIZE`) is set to 251, the actual used packet size is still 27 bytes. 

For some reason, the larger packet size is only applied for the RX MTU, but not for the TX MTU:
[image]
[image]

I have tried to find the part of the controller initialization where these values are applied, but so far I did not succeed to find the reason. Could maybe point me in a direction where to look in particular?

## Details
hw: 2x `nrf52dk`
NimBLE (fb7651b) @ RIOT (0e736b8)
node 1: running `RIOT/tests/nimble_l2cap`
node 2: running `RIOT/tests/nimble_l2cap_server`
Sending data using the `flood [chunksize] [num]` shell command on node 1.

NimBLE config:
```
BLE_LL_CFG_FEAT_DATA_LEN_EXT 1
BLE_LL_MAX_PKT_SIZE (251)
BLE_LL_SUPP_MAX_RX_BYTES = BLE_LL_MAX_PKT_SIZE
BLE_LL_SUPP_MAX_TX_BYTES = BLE_LL_MAX_PKT_SIZE
BLE_L2CAP_COC_MPS = 250
```

haukepetersen, 2019-11-15:
Found it. Apparently there is the additional configuration option `BLE_LL_CONN_INIT_MAX_TX_BYTES` which is currently set per default set to 27. This value then restricts the link layers maximum TX packet size to that value, independent of the values chosen for `BLE_LL_SUPP_MAX_TX_BYTES`.

Before closing this issue, could someone explain the difference and how these values should be used?

sjanc, 2019-11-18:
BLE_LL_CONN_INIT_MAX_TX_BYTES defines value that controller will try to get automatically when connected. If this is set to 27 controller will not do this on its own and leave configuration up to host.

haukepetersen, 2019-11-18:
I see, thats what I figured by now, but its nice to have it verified. So i consider this issue solved.

PR #654 (2019-11-19): npl/riot/syscfg: fix BLE_LL_CONN_INIT_MAX_TX_BYTES
Currently, the default value of `BLE_LL_CONN_INIT_MAX_TX_BYTES` is set to 27, preventing the LL data length extension to work properly in RIOT (TX octets always set to 27). With this fix the controller is applying the desired TX octets values as defined by `BLE_LL_MAX_PKT_SIZE`.

fixes #649 for RIOT.
closed 2019-11-18 by haukepetersen