apache/mynewt-nimble #656: Assert in `ble_ll_scan_start`

simonratner, 2019-11-16:
Seeing an assert trigger under load in `ble_ll_scan_start`, line 1010:
```
BLE_LL_ASSERT(scansm->scan_rsp_pending == 0);
```
[image]

We are initiating the scan proc periodically, with duration of 1500ms, then restarting it once it completes.

Any help appreciated.

simonratner, 2019-11-16:
Running commit 19d7ebf.

sjanc, 2019-11-16:
Hmm we've been seeing this assert on UPF and hoped that it was fixed but apparently I've missed clearing that also on duration timer fire,  could you try this?
https://github.com/apache/mynewt-nimble/pull/657

lance-proxy, 2019-11-18:
@sjanc, thanks for your patch. Im working with @simonratner on this. We have updated some of our test devices and have yet to see this assertion yet. I will post here straight away if I observe this assertion.

sjanc, 2019-11-22:
any update on this? should we keep this stil open for now?

lance-proxy, 2019-11-22:
Hi @sjanc sorry about the late reply. We have not been hitting this assertion in `ble_ll_scan_start` since adding your patch. I see #657 is merged so I think we can close this issue.

closed 2019-11-25 by sjanc