apache/mynewt-nimble #664: L2CAP COC: We should get an error when trying to create more servers than configured

michal-narajowski, 2019-11-19:
If we try to create a server for more connections than we support (e.g. we support 1 connection and then we try to create a second server) we do not return an error. The error is returned only when the Client tries to connect to us. Maybe we should throw an error when creating the server to save time.

rymanluk, 2019-11-20:
There should be possible to create many CoC servers and then use them on the same ACL connection. Is there an issue with that?

For sure, number of supported acl connections is not related to number of supported CoC...well not counting that you need at least one acl :)

rymanluk, 2019-12-30:
@michal-narajowski do you think we still need this PR or can we close it?

michal-narajowski, 2019-12-31:
This issue appears when we create BLE_L2CAP_COC_MAX_NUM channels as a client and then we create an L2CAP server. When a peer tries to connect to it we will return an error because we will not be able to allocate a new channel (limit of channels reached). We can create a server successfully because this API uses `ble_l2cap_coc_srv_pool` mempool. Then when a peer connects to this server we try to allocate a channel from `ble_l2cap_chan_pool` mempool and fail. 

I'm not saying this is a bug per se, more of a usability issue.

KKopyscinski, 2020-08-03:
I think this issue no longer exists - size of `ble_l2cap_coc_srv_pool` is limited by BLE_L2CAP_COC_MAX_NUM in `ble_l2cap_coc`, method `ble_l2cap_coc_init`. If we try to create more servers, we get error 0x06 - BLE_HS_ENOMEM.