apache/mynewt-nimble #668: Duration misinterpreted by ble_gap_ext_adv_start

simonratner, 2019-11-19:
Docs for `ble_gap_ext_adv_start` say duration is in ms, which is the same as for the legacy `ble_gap_adv_start` function, however this duration param appears to be passed unmodified to the HCI command, and LL interprets this duration in units of 10ms. 

Compare:
https://github.com/apache/mynewt-nimble/blob/master/nimble/host/src/ble_gap.c#L2653
https://github.com/apache/mynewt-nimble/blob/master/nimble/host/src/ble_gap.c#L2711
https://github.com/apache/mynewt-nimble/blob/master/nimble/controller/src/ble_ll_adv.c#L3418

Documentation is here:
https://github.com/apache/mynewt-nimble/blob/master/nimble/host/include/host/ble_gap.h#L1204

I think the param should be renamed `duration_ms` and docs changed to match the legacy function, which was the intent, however that's obviously a breaking change. Thoughts on how to proceed?

andrzej-kaczmarek, 2019-11-19:
Seems like description is modified just c&p from legacy adv API and note about milliseconds was left there by mistake. In general all APIs use the same units as corresponding HCI commands so imo we should leave this parameter as-is and only fix description.

simonratner, 2019-11-19:
Code at https://github.com/apache/mynewt-nimble/blob/master/nimble/host/src/ble_gap.c#L2653 needs to be fixed also then.

simonratner, 2019-11-19:
Ok, will work on assumption that the documentation is wrong and that the units will remain as 10ms, thanks.

andrzej-kaczmarek, 2019-11-19:
^^ @sjanc is my assumption correct or it was supposed to be ms?

sjanc, 2019-11-19:
yeap, the idea was to have this all match HCI params since otherwise we would have to play trick with rounding to HCI units.

sjanc, 2019-11-20:
https://github.com/apache/mynewt-nimble/pull/669

sjanc, 2019-11-22:
669 is merged so I think this can be closed too?

closed 2019-12-20 by sjanc