apache/mynewt-nimble #685: ble_phy_isr in nrf52 series

JaydenH215, 2019-11-30:
Hi, guys
   How should I understand the following operation, i think there is no correspondence between irq and NRF_RADIO-> SHORTS after i read spec about nrf52832.

    /* Ensures IRQ is cleared */
    irq_en = NRF_RADIO->SHORTS;

Thanks.

wes3, 2019-11-30:
It is not the greatest comment in the world to be sure, but this comes from the following section in the spec: nrf52832_PS v1.4. Section 15.8.1 Interrupt clearing.

Important: To avoid an interrupt reoccurring before a new event has come, the program should perform a read from one of the peripheral registers, for example, the event register that has been cleared, or the INTENCLR register that has been used to disable the interrupt.

JaydenH215, 2019-12-01:
Thanks, i got it.

closed 2019-12-01 by JaydenH215