apache/mynewt-nimble #687: Return code from ble_hs_hci_cmd_tx is HEX207, while it should be a BLE_HS_E... type of error

robkers1993, 2019-12-03:
Hi all!

Im porting the driver part of the Nimble stack to my own platform. The whole thing is work in progress and i'm trying to run the stack. The problem that I run into right now, is running the Advertisement example where 

Now, in the call ble_hs_hci_cmd_tx(..) at line 338 "rc = ack.bha_status". This value is set to DEC519, which translates to 0x207. However, if I look at the definition of the ACK type, i see in the comment that I have to interpret this error as /* A BLE_HS_E<...> error; NOT a naked HCI code. */

Now, looking at https://mynewt.apache.org/latest/network/docs/ble_hs/ble_hs_return_codes.html , there is no code BLE_HS_E... with the value 0x207. For sure, a normal HCI can take that value, but it should not be interpreted that way.

Q1: How should I terpret this rc? As a HS or a HCI rc?
Q2: Why is the rc set to a value it could in theory not take? The RC is incompatible with ack.bha_status.

Cheers,
Rob

andrzej-kaczmarek, 2019-12-03:
error codes with base `0x200` are HCI errors so `0x207` is a HCI error code 7 (memory capacity exceeded)

robkers1993, 2019-12-03:
I see that my question was not completely clear;

Do I have to trust the comment behind the struct definition of the ack struct saying

bha_status /* A BLE_HS_E<...> error; NOT a naked HCI code. */

This is since there is a HCI code set to that field, although the documentation says it should not.

andrzej-kaczmarek, 2019-12-03:
comment is correct - this is not a raw HCI code, but a HCI code converted to BLE_HS error so it can be returned to application and will not be confused with "standard" BLE_HS error. each raw HCI code has corresponding BLE_HS error which is basically HCI code + 0x200 and this is what is set in `bha_status`

robkers1993, 2019-12-03:
Ah, I see, Missed the conversion step, sorry. Thanks for the quick reply!

closed 2019-12-09 by robkers1993