apache/mynewt-nimble #689: run NimBLE on embeded linux platform

mirandw, 2019-12-05:
I am porting Nimble on embeded linux platform(mips arch/ linux-4.9/realtek 8723du bt module).
After some modification, nimble-linux runs smoothly.
But the size of Nimble and the memory consumed by Nimble is much larger than acclaimed by official .
Size of nimble-linux: 1.1MB
Memory consumed by nimble-linux: 1.7MB
acclaimed by official:
4.5KB RAM，69KB Flash
I wonder why NiBLE consumes so much memory?
Have anyone run Nimble on Linux? How much memory it consume?

andrzej-kaczmarek, 2019-12-05:
You can't expect NimBLE running as a standalone process to be nowhere near quoted numbers because it's not NimBLE itself which contributes to image size and memory usage. When linked as executable binary is has multiple system libraries linked and those have immediate effect on both image size and (virtual) memory usage, as on any other Linux binary.

BTW, one simple thing you can do to decrease image size is to strip binary which removes all debug symbols (there are plenty of them).

mirandw, 2019-12-09:
thank you, i will try again

closed 2021-05-27 by sjanc