apache/mynewt-nimble #694: Assertion `!ble_hs_locked_by_cur_task()' failed on Linux Platform

mirandw, 2019-12-09:
When I run NimBLE on linux platform, and enables MYNEWT_VAL_BLE_HS_DEBUG
the error occurs:
nimble/host/src/ble_gap.c: 607: ble_gap_call_event_cb: Assertion `!ble_hs_locked_by_cur_task()' failed.
follow the function ble_hs_lock() --->BLE_HS_DBG_ASSERT(!ble_hs_locked_by_cur_task())--->
ble_hs_locked_by_cur_task() {
#if MYNEWT
    struct os_task *owner;

    if (!ble_npl_os_started()) {
        return ble_hs_dbg_mutex_locked;
    }

    owner = ble_hs_mutex.mu.mu_owner;
    return owner != NULL && owner == os_sched_get_current_task();
#else
    return 1;
#endif
when under Linux, the function returns 1 always.
so BLE_HS_DBG_ASSERT(!ble_hs_locked_by_cur_task()) ≈assert(0)
it fails definitely.
Any one come with this question?

v01d, 2020-09-27:
I found the same problem. I had to disable debug assertions to work around this.

artyeffem, 2021-02-17:
I added this to the `#else` (i.e. Linux) section
`return (ble_hs_mutex_locked && ble_hs_task_handle == ble_npl_get_current_task_id());`

Haven't hit this assert for months until passing through the `ble_store_util_status` handler that calls `ble_gap_unpair_oldest_peer`.
Should there be `ble_hs_lock/unlock` around this unpair operation?

sjanc, 2026-10-30:
fixed in master

closed 2026-10-30 by sjanc