apache/mynewt-nimble #703: Beacons Picked up by Iphone but not Android

robkers1993, 2019-12-16:
Hi All!

TLDR: Why can I not receive this packet on Android devices (4 devices tested), but Iphones are fine (two Iphones tested)? Did I set a super obvious setting wrong?

20 19 55 11 22 33 44 55 02 01 06 03 03 11 18 07 09 6e 69 6d 62 6c 65 02 0a 0c 
20195511223344550201060303111807096e696d626c65020a0c

I'm running the BLE peripheral example in the apps folder. CRC is 3 bytes, whitening seed looks allright, 

Longer situation sketch:

I'm porting my radio driver to the Nimble Radio Driver (so adding it next to native, nrf51,nrf52). I have a stable beaconing environment, so I would like to connect to a device and get interact with a service. My Iphone is able to receive beacons. The problem starts when I want to connect to it. I have never received a SCAN REQ or CONN REQ from my phone. I've printed all incoming data with a filter on pdu_type on these two types of messages (0x03 and 0x05). In an effort to see whether or not my phone is functioning correctly I tested the same scenario with a borowed iphone 11 pro max (yeezzz, these things are unwieldy coming from an Iphone SE.... anyhow...). This phone picked it up with no effort.
Then tested a lot of other hypotheses, without any luck of getting a SCAN REQ or CONN REQ. Then, I grabbed 4 Android devices from my colleagues and tested the same thing. BANG: devices do not show up using something like NRF Connect. 
Also, grabbed a ESP32 and ran the BLE scan example. My Nimble device is picked up without an effort. 

I hope anyone out there has an idea about this. I've got the feeling that one of the fields/flags are set incorrectly. Below the settings from the "bleperiph" example. 

Cheers!

__________________
    fields.flags = BLE_HS_ADV_F_DISC_GEN |BLE_HS_ADV_F_BREDR_UNSUP;

    fields.tx_pwr_lvl_is_present = 1;
    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;

    name = ble_svc_gap_device_name();
    fields.name = (uint8_t *)name;
    fields.name_len = strlen(name);
    fields.name_is_complete = 1;

    fields.uuids16 = (ble_uuid16_t[]){
        BLE_UUID16_INIT(GATT_SVR_SVC_ALERT_UUID)
    };
    fields.num_uuids16 = 1;
    fields.uuids16_is_complete = 1;

    adv_params.conn_mode = BLE_GAP_CONN_MODE_UND;
    adv_params.disc_mode = BLE_GAP_DISC_MODE_LTD;

sjanc, 2019-12-19:
SCAN_REQ and CONNECTION_REQ should be expected after 150us (T_IFS) +/- 2 us. So your radio needs to be able to go from TX to RX on time.

andrzej-kaczmarek, 2019-12-19:
IIRC Android will not show devices which do not reply to SCAN_REQ so this is why your device is visible in ESP32 but not on Android. You can change advertising mode to non-connectable and non-scannable (i.e. an actual beacon) to verify this.

robkers1993, 2019-12-19:
I'm using a SX1280, which should be capable of that. Also, I did some time related measurements to verify I'm setting the radio quickly enough into Rx mode. I'm debugging this atm, question related to that belong on the sx1280 forum. 

I'm still not able to recieve my data with an Android phone using the Nordic Connect toolbox. Next to a phone, i'm using a nrf51 dev kit as a sniffer. Now Wireshark can show me the packets between my phone and sx1280. I can verify that a correct (enought) ADV_IND is transmitted from the sx1280 and my iPhone responds quickly enough (150 us) with a CONN_REQ.

robkers1993, 2019-12-19:
@andrzej-kaczmarek , That indeed did the trick. 

I'v set

    fields.flags =  BLE_HS_ADV_F_DISC_LTD|
                    BLE_HS_ADV_F_DISC_GEN |
                    BLE_HS_ADV_F_BREDR_UNSUP;

    /* Begin advertising. */
    memset(&adv_params, 0, sizeof adv_params);
    adv_params.conn_mode = BLE_GAP_CONN_MODE_NON;
    adv_params.disc_mode = BLE_GAP_DISC_MODE_NON;

andrzej-kaczmarek, 2019-12-19:
I've just quickly looked over SX1280 datasheet and there are at least few problems running it with NimBLE. The main problem is that this is an external transceiver so after each RX and before each TX you have to copy data over SPI which adds latency when compared with integrated RF. Also seems like TX/RX starts automatically as soon as SPI command is executed (+ramp up time) so the timing is effectively determined by SPI transmission time - you need to carefully calculate all timings here and take possible latency into account on each step. This means that you will absolutely need logic analyzer to debug and fine-tune this since SPI and IRQ lines will be basically your reference for timing.

In theory, if your host MCU is fast enough and SPI runs at 18MHz you should be able to program TX-RX and RX-TX transitions on time and have basic advertising and connection working. However, for RX-TX transition since you can start processing packet only after it was received completely and copied to host, it means you have less time to process it than in usual NimBLE configuration where some initial checks are done just after header was received so I expect timing will be really tight here. Based on our experience most problems we had with timing was to be able to copy PDU on time and this was with just a simple memcpy to RAM. The fastest MCU we used however was CM4F (nRF52840) so perhaps you have something faster there.

Finally, there is some "accelerator" for RX-TX transition built-in but it's useless for NimBLE since you would need to program TX packet while still receiving and NimBLE does not know what to transmit before received PDU is processed.

sjanc, 2021-05-27:
please reopen if you would need any further comments on this

closed 2021-05-27 by sjanc