apache/mynewt-nimble #712: A case in ble_ll_sched_adv_reschedule

JaydenH215, 2020-01-01:


wnnwoo, 2020-01-11:
Yes, it may overlap the next_sch

and the procedure will judge whether it is overlapped it in the next iteration in this while loop, the procedure will go and find the right position until the end of the Q.

sjanc, 2020-01-13:
Hi,

Since our documentation needs improvements, it would be great if you consider updating it with your notes and send PR :-)

JaydenH215, 2020-01-13:
> Yes, it may overlap the next_sch
> 
> and the procedure will judge whether it is overlapped it in the next iteration in this while loop, the procedure will go and find the right position until the end of the Q.
> 
> U can refer this doc.
> https://wnnwoo.github.io/2020/01/05/nimble-scheduler/

```
    if (entry == end_overlap) {
        rand_ticks = (orig_start + max_delay_ticks) - sch->start_time;
        if (rand_ticks > max_delay_ticks) {
            /* No place for advertisement. */
            rc = -1;
        } else {
            if (next_sch == NULL) {
                TAILQ_INSERT_TAIL(&g_ble_ll_sched_q, sch, link);
            } else {
                TAILQ_INSERT_BEFORE(next_sch, sch, link);
            }
        }
        break;
    }
```

But the latter condition should be true, it will 'break' and will not enter the while again.
I have read your blog and thank you for sharing. I am also learning nimble and hope to make friends with you. 

[https://jaydenh215.github.io/](https://jaydenh215.github.io/)

wnnwoo, 2020-01-13:
Dose my blog resole ur question?

sjanc, 2021-05-27:
reopen if this is stil an issue

closed 2021-05-27 by sjanc