apache/mynewt-nimble #736: Issue in init and deInit of BLE and understanding of code

umer-ilyas, 2020-01-22:
sometimes i am having this problem after i call `nimble_port_stop();  nimble_port_deinit();` 

`assertion "ret == pdPASS" failed: file "F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c", line 291, function: npl_freertos_sem_release`

I am using esp-idfv3.3.1 with ESP32-WROOM
and tested with both nimble-1.1.0-idfv3.3 and nimble-1.2.0-idf

here is full assertion log

```
assertion "ret == pdPASS" failed: file "F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/npl/freertos/src/npl_o
s_freertos.c", line 291, function: npl_freertos_sem_release
abort() was called at PC 0x4010fd17 on core 0
0x4010fd17: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

ELF file SHA256: a82d01ccf69a5c9f5068f08aad415292749fc0a546dfafccaf1963fd72c2cfba

Backtrace: 0x4008e978:0x3fff0c80 0x4008ebc5:0x3fff0ca0 0x4010fd17:0x3fff0cc0 0x4017db73:0x3fff0cf0 0x4017cd76:0x3fff0d20 0x4017590a:0x3fff0d40 0x40175922:0x3fff0d60 0x4017596e:0x3fff0d80 0x4017059e:0x3fff0da0 0x4017142c:0x3fff0dc0 0x40171454:0x3fff0e20 0x40176da2:0x3fff0
e80 0x40176dd2:0x3fff0eb0 0x40173d4d:0x3fff0ed0 0x4017cdce:0x3fff0ef0 0x40137ab8:0x3fff0f10 0x40097cc9:0x3fff0f30
0x4008e978: invoke_abort at F:/msys32/home/umeri/esp/esp-idf/components/esp32/panic.c:715

0x4008ebc5: abort at F:/msys32/home/umeri/esp/esp-idf/components/esp32/panic.c:715

0x4010fd17: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4017db73: npl_freertos_sem_release at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:369

0x4017cd76: ble_npl_sem_release at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:202
 (inlined by) ble_hs_stop_cb at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/nimble/src/nimble_port.c:94

0x4017590a: ble_hs_stop_done at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_hs_stop.c:57 (discriminator 3)

0x40175922: ble_hs_stop_terminate_next_conn at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_hs_stop.c:75

0x4017596e: ble_hs_stop_gap_event at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_hs_stop.c:117

0x4017059e: ble_gap_event_listener_call at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_gap.c:1682

0x4017142c: ble_gap_conn_broken at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_gap.c:1682

0x40171454: ble_gap_rx_disconn_complete at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_gap.c:1682

0x40176da2: ble_hs_hci_evt_disconn_complete at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:167

0x40176dd2: ble_hs_hci_evt_process at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:800

0x40173d4d: ble_hs_event_rx_hci_ev at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:109

0x4017cdce: ble_npl_event_run at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121
 (inlined by) nimble_port_run at F:/msys32/home/umeri/esp/esp-idf/components/nimble/nimble/porting/nimble/src/nimble_port.c:81

0x40137ab8: bleprph_host_task at F:/msys32/home/umeri/esp/cowlar-finger-print-scanner-esp32/components/iot-core/ble/app_ble.c:579

0x40097cc9: vPortTaskWrapper at F:/msys32/home/umeri/esp/esp-idf/components/freertos/port.c:403
```

i have problem in understanding this part of code 
this function never returns because of `SLIST_FOREACH ` keep calling the listener callback function and i could not found a code which remove the callback function from `slist`
```
/**
 * Called when a stop procedure has completed.
 */
static void
ble_hs_stop_done(int status)
{
    struct ble_hs_stop_listener_slist slist;
    struct ble_hs_stop_listener *listener;

    ble_hs_lock();

    ble_gap_event_listener_unregister(&ble_hs_stop_gap_listener);

    slist = ble_hs_stop_listeners;
    SLIST_INIT(&ble_hs_stop_listeners);

    ble_hs_enabled_state = BLE_HS_ENABLED_STATE_OFF;

    ble_hs_unlock();

    SLIST_FOREACH(listener, &slist, link) {
        listener->fn(status, listener->arg);
    }
}
```

and sometime in future    ` in nimble_port.c @ 127 ble_npl_sem_deinit(&ble_hs_stop_sem);` semaphore gets deinit and this function 
```
ble_hs_stop_cb(int status, void *arg)
{
    ble_npl_sem_release(&ble_hs_stop_sem);
}
```
is still trying to release it and there rtos issue assert

tennten10, 2021-06-17:
I'm having a similar issue when trying to deinit on esp32. I've traced it to ble_npl_sem_pend() inside the nimble_port_stop() function and it never gets past that. Were you able to resolve it?

felixcollins, 2023-03-22:
Has this been resolved? I'm on ESP32 and nimble_port_deinit(); seems to hang. It seems like basic functionality to be able to start/stop the BTLE.