apache/mynewt-nimble #749: HCI BCSP protocol support

dmastain, 2020-02-04:
I'm wondering if there is any plan to support the BCSP HCI over UART standard. I'm in a situation where I would like to use a UART to send HCI commands from Host to Controller but don't have enough pins available for hardware flow control and BCSP seems like a promising solution but it doesn't appear to be currently available in the stack.

Are there currently any other solutions around flow control?

rymanluk, 2020-04-24:
As far as I know BCSP is (BlueCore Serial Protocol) and since it is not Bluetooth SIG, we don't have motivation for that. However, if somebody I believe will implement it we will accept patches :)

LuoZhongYao, 2020-05-11:
H5 is the standard protocol of Bluetooth SIG, he is very similar to BCSP, but H5 is simpler

closed 2020-07-24 by rymanluk