apache/mynewt-nimble #785: [npl] linux porting layer unit tests fail

turon, 2020-03-26:
```
~/mynewt-nimble/porting/npl/linux/test$ make test
./test_npl_task.exe
All tests passed
./test_npl_callout.exe
All tests passed
./test_npl_eventq.exe
All tests passed
./test_npl_sem.exe
make: *** [Makefile:94: test] Segmentation fault
```

It may be good to add these tests as a github action that gates pull requests.

turon, 2020-03-26:
Looks like this can be fixed by replacing main loop in [`test_os_sem.c`](https://github.com/apache/mynewt-nimble/blob/master/porting/npl/linux/test/test_npl_sem.c#L150):

BEFORE:
```
   while (1)
    {
        ble_npl_eventq_run(ble_npl_eventq_dflt_get());
    }
```

AFTER:
```
    while (1) {}
```

sjanc, 2020-04-08:
hmm works for me for both 1.3 and master
[janc@ix test]$ make test
./test_npl_task.exe
All tests passed
./test_npl_callout.exe
All tests passed
./test_npl_eventq.exe
All tests passed
./test_npl_sem.exe
All tests passed

closed 2020-07-24 by rymanluk