apache/mynewt-nimble #788: nimble/store: `ble_gap_unpair_oldest_peer` in `ble_store_util_status_rr`(store callback) can cause infinite loop for CCCDs

prasad-alatkar, 2020-03-30:
In cases where CCCDs are exceeding `MYNEWT_VAL(BLE_STORE_MAX_CCCDS)` (overflow event), we can end up in continuous while loop. 

Whenever `BLE_STORE_EVENT_OVERFLOW` is called in store callback, if we do not have any previously bonded peer, `ble_gap_unpair_oldest_peer` will return `0` without deleting any CCCD. This in turn results in continuous while loop in `ble_store_write` as it will continuously end up calling `BLE_HS_ESTORE_CAP` evwnt.

h2zero, 2020-03-31:
I found this as well, currently testing a fix in my own fork. works for my purposes.
You can see it [here](https://github.com/h2zero/NimBLE-Arduino/blob/bae2eb26d0128d649eb854bc23a4846d7d7581fc/src/nimble/host/src/ble_gap.c#L5305)

prasad-alatkar, 2020-04-06:
@h2zero your proposed solution may avoid this issue when "Bonding" is not enabled, but the issue will eventually pop-up in case of bonding is enabled (i.e. `ble_hs_cfg.sm_bonding = 1`).

h2zero, 2020-04-06:
Yes, you are correct. It was just a quick patch to get by while developing the cpp library to avoid that error. Didn’t want to do any big changes to the NimBLE core code yet.

I do not think we should store cccd data if not bonding though, maybe a combination of my fix and a proper round robin cccd store patch?

rymanluk, 2020-04-24:
This is fixed. Closing.

closed 2020-04-24 by rymanluk