apache/mynewt-nimble #789: How to use ble_gap_connect?

joonazan, 2020-03-31:
I am using NimBLE with esp-idf. I would like to set the connection interval but I can't figure out how.

ble_gap_connect lets you set the interval but it seems that ble_gap_adv_start calls it for you.

What is the most sane way of doing this?

sjanc, 2020-04-20:
Hi, 

You provide connection parameters as  "const struct ble_gap_conn_params *params" parameter.
On connection controller will choose interval based on itvl_min and itvl_max provided.

Note that connection interval has nothing to do with advertising (and scan) interval.

joonazan, 2020-04-22:
I know that. I want a frequent connection interval so that I can transfer more and with less latency.

But I don't think I can call ble_gap_connect. It isn't called anywhere in the esp-idf examples.

I start advertising with ble_gap_adv_start and then I get a callback after a connection is created. So I guess that some code is calling ble_gap_connect for me.

Link to my code: https://github.com/joonazan/nina-fast-bluetooth/blob/master/extras/nina-src/main/ble_connection.c

rymanluk, 2020-04-24:
So you are peripheral. You can as central to change the parameter and he might accept it.

You should look at `ble_gap_update_param()`  http://mynewt.apache.org/latest/network/ble_hs/ble_gap.html

closed 2020-04-24 by rymanluk