apache/mynewt-nimble #793: min\max_ce_len, how is it used internally?

h2zero, 2020-04-07:
I've been trying to track down errors when acting as a client connecting to multiple peripheral devices. What I have experienced so far is when the connection intervals are short, approx. 20ms, 2 connections are fine but once connected to a 3rd it causes an error and aborts on the esp32.

I am finding by setting the min_ce_len to half of the default initial value, 0x0010 to 0x0008, the errors dissipate. 

So what I'm asking is does the min_ce_len specify a fixed time slot that NimBLE reserves for that particular connection, regardless of any communication between the devices?

If so would that mean in this instance with 2 devices connected, each with 20ms connection intervals and 10ms minimum connection event lengths that there would be no time available to support a 3rd connection?

andrzej-kaczmarek, 2020-04-07:
NimBLE controller does not use min/max CE length parameters for scheduling but it does validate them (likely there was intention to use them, but it was never implemented).

There's syscfg settings which affects connection event length, namely `BLE_LL_CONN_INIT_SLOTS`, which sets minimum number of slots allocated for each connection event. It usually does not affect limit the number of concurrent connections controller is able to support since it will just skip connection events it cannot schedule. When there are 2 colliding connection events, the least recently scheduled connection has priority (the other one is skipped and that connection moves to next connection event). So with multiple connections with short connection interval just make sure supervision timeouts are long enough so controller can skip connection events without timeouts happening.

h2zero, 2020-04-07:
Thanks, the helps a lot 😃

h2zero, 2020-04-19:
Reopening this because I have found that when those parameters are set on a device with a separate controller, such as the esp32, it may use them as provided. This explains why changing them in my testing proved to have positive results.

Would there be any issues for your controller if I were to submit a PR that sets these defines:
```
#define BLE_GAP_INITIAL_CONN_MIN_CE_LEN     0x0010
#define BLE_GAP_INITIAL_CONN_MAX_CE_LEN     0x0300
```
To:
```
#define BLE_GAP_INITIAL_CONN_MIN_CE_LEN     0x0000
#define BLE_GAP_INITIAL_CONN_MAX_CE_LEN     0x0000
```
The performance on the esp32 with these settings is absolutely fantastic, before this change the response was subpar. But if it causes issues outside the esp32 framework I will submit the proposal there perhaps.

andrzej-kaczmarek, 2020-04-24:
@h2zero yes, it makes sense to set default values to 0. I do not really know why current values were set, but since NimBLE does not use them anyway and they can be overridden at any time it's better to have them at "neutral" values.

h2zero, 2020-04-24:
@andrzej-kaczmarek Thanks, I just posted the PR, #808.

h2zero, 2020-04-27:
PR merged, closing.

closed 2020-04-27 by h2zero