apache/mynewt-nimble #817: Central/Peripheral + BLE Mesh at the same time?

zacwbond, 2020-05-15:
Is it possible to run BLE mesh in addition to standard BLE?  

I would like to implement mesh alongside existing BLE central and peripheral functions.  I have gotten the mesh to work, but the central and peripheral code is broken because the `ble_gap_adv_start()` always returns `EALREADY`, since, of course, the mesh code is advertising.

I see that if extended advertisements are enabled, then the mesh-related advertising looks to be confined to the two upper-most advertising instances, and at I could simply run my current code using a separate instance.  Unfortunately, I am on the ESP32 platform, which is a 5.0 device that doesn't actually support a bunch of 5.0 features, such as extended advertisements, so that solution won't work.

Right now my only idea is to periodically call `bt_mesh_suspend()` and `bt_mesh_resume()`, with brief periods for my own advertising in between.  I'd be interested to know if this should work or, if not, what my other options are.  Maybe I'm missing something obvious.

rymanluk, 2020-07-24:
@zacwbond 

Imho toggling advertising between mesh and standard BLE  using `bt_mesh_suspend()/resume()` might be risky. But I can't see other option.

sjanc, 2022-04-13:
should be possible with extended advertising instances,  reopen if needed

closed 2022-04-13 by sjanc