apache/mynewt-nimble #824: L2CAP COC: nimble@mynewt crashes when flooding a channel

haukepetersen, 2020-05-27:
## Context
I am still trying to understand the behavior described in #818. To make sure this is not a RIOT or a porting issue, I ported the benchmark application to mynewt (-> https://github.com/haukepetersen/riotsandbox/tree/master/tests/mynewt_nimble_l2cap_coc_flooding)
This is the first time I actually did something with mynewt and I am still learning - so please excuse the project structure and code maturity :-)

## Problem
When flooding an L2CAP COC link between two nodes for a long enough amount of time (~some seconds), the transmitting node will crash (reboot).
By e.g. trying to send 10k chunks of 100byte (`flood 100 10000`), the `source` node will reboot after a random time, mostly something around ~15-20 seconds.

## Setup
Two `nrf52dk` nodes on my desk, one flashed with the `source` application (target `source`), one with the `sink` (target `sink`) application.
The nodes open a BLE connection automatically, and once connected also open a custom L2CAP connection oriented channel.
Then, once connected, simply call the `flood` shell command on the `source` node.

Is anyone able to reproduce this behavior? And does anyone have an idea where to look further? I am a little stuck here...

rymanluk, 2020-07-06:
Sorry for late reply, I had no time to look into it till now.

I could reproduce it and this is watchdog  restarting source. Solution for your issue is to use `os_sem` instead of `os_mutex`. If I udenrstand corretly  you want to wait here: https://github.com/haukepetersen/riotsandbox/blob/master/tests/mynewt_nimble_l2cap_coc_flooding/apps/source/src/main.c#L64  untli UNSTALLED will be received.

It works for me when I changed that.

haukepetersen, 2020-08-13:
No worries, same sorry here :)

>  If I understand corretly you want to wait here

Yes, that is correct. The aim is to allow other, possibly lower prio tasks to work while the sending task is waiting for L2CAP to sent the previous packet. Using a semaphore/mutes here seemed the obvious choice, right? Or do you have an idea for something better suited for synchronizing the sender task?

Will also try to run my code using a semaphore and report back with my results.

haukepetersen, 2020-08-24:
Confirmed: using os_sem solves this issue. Interesting system design to have the watchdog triggering for waiting mutexes but not for semaphores :-) Maybe that could be added to the doc pages?!

Anyhow, everything is in order now, so I consider this issue solved.

closed 2020-08-24 by haukepetersen

andrzej-kaczmarek, 2020-08-24:
> Confirmed: using os_sem solves this issue. Interesting system design to have the watchdog triggering for waiting mutexes but not for semaphores :-) Maybe that could be added to the doc pages?!

that's not the problem with mutex itself. sample code acquires mutex in on task (flooder) and releases it in another (main). this means an attempt to release mutex always returns an error since task which tries to release mutex does not own it, while an attempt to acquire mutex will not block since mutexes are recursive so the same task can acquire a mutex more than once. so effectively sample code spins in `_send` which triggers watchdog.