apache/mynewt-nimble #834: Assert at ble_ll_scan_aux_data_unref

lance-proxy, 2020-06-21:
Seeing an assert under load in ble_ll_scan_aux_data_unref in `ble_ll_scan.c` at line 1079: nimble/blob/master/nimble/controller/src/ble_ll_scan.c#L1079, running latest `master`, commit: https://github.com/apache/mynewt-nimble/commit/cf8e8125fd2428adffd78ef0ea03fdae4faec88c

```
BLE_LL_ASSERT(aux_data);
````

[image]

We have `BLE_EXT_ADV` enabled and are initiating the scan proc periodically, with duration of 1500ms, then restarting it once it completes.

lance-proxy, 2020-06-21:
@sjanc @andrzej-kaczmarek this looks like the same assertion that was attempted to be fixed by this PR: https://github.com/apache/mynewt-nimble/pull/748 . Any help in fixing this would be appreciate.

lance-proxy, 2020-06-21:
(I have not yet been able to repro this on the bench with the debugger attached but as soon as I hit it, I will upload the full gdb backtrace).

andrzej-kaczmarek, 2020-06-22:
@lance-proxy I've seen this assert a few times when running qualification testcases but also was not able to catch it on a debugger. there was a bit of refactoring done in scanner so I guess I missed a corner case somewhere, but don't have clue so far. backtrace would definitely help a lot.

lance-proxy, 2020-06-29:
@andrzej-kaczmarek @rymanluk  I finally caught a test device in the debugger. See attached gdb backtrace:

```
(gdb) bt
#0  hal_debug_break () at repos/apache-mynewt-core/hw/mcu/nordic/nrf52xxx/include/mcu/cortex_m4.h:37
#1  ble_ll_scan_aux_data_unref (aux_data=0x0) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_scan.c:1079
#2  0x000309a8 in ble_ll_scan_rx_pkt_in_on_aux (addrd=0x2001396c <g_ble_ll_stack+380>, hdr=0x2000d694 <os_msys_1_data+1168>, om=0x2000d67c <os_msys_1_data+1144>, pdu_type=<optimized out>)
    at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_scan.c:3183
#3  ble_ll_scan_rx_pkt_in (ptype=<optimized out>, om=om@entry=0x2000d67c <os_msys_1_data+1144>, hdr=0x2000d694 <os_msys_1_data+1168>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_scan.c:3207
#4  0x00029896 in ble_ll_rx_pkt_in () at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:834
#5  ble_ll_event_rx_pkt (ev=<optimized out>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:1161
#6  0x000297ae in ble_npl_event_run (ev=<optimized out>) at repos/apache-mynewt-nimble/porting/npl/mynewt/include/nimble/nimble_npl_os.h:116
#7  ble_ll_task (arg=<optimized out>) at repos/apache-mynewt-nimble/nimble/controller/src/ble_ll.c:1211
#8  0x000163d2 in os_time_ms_to_ticks (ms=536951248, out_ticks=0x2001398c <g_ble_ll_stack+412>) at repos/apache-mynewt-core/kernel/os/src/os_time.c:346
#9  0x00000000 in ?? ()
```

Some LL global variable inspections:
```
(gdb) p g_ble_phy_data
$1 = {phy_stats_initialized = 1 '\001', phy_txpwr_dbm = 0 '\000', phy_chan = 37 '%', phy_state = 0 '\000', phy_transition = 2 '\002', phy_transition_late = 0 '\000', phy_rx_started = 1 '\001', 
  phy_encrypted = 0 '\000', phy_privacy = 0 '\000', phy_tx_pyld_len = 12 '\f', phy_cur_phy_mode = 1 '\001', phy_tx_phy_mode = 1 '\001', phy_rx_phy_mode = 1 '\001', phy_bcc_offset = 0 '\000', 
  rx_pwr_compensation = 0 '\000', phy_aar_scratch = 0, phy_access_address = 2391391958, rxhdr = {{rxinfo = {flags = 2050, channel = 37 '%', handle = 0 '\000', rssi = -92 '\244', phy = 1 '\001', 
        phy_mode = 1 '\001', user_data = 0x0}, txinfo = {flags = 2 '\002', offset = 8 '\b', pyld_len = 37 '%', hdr_byte = 0 '\000'}}, beg_cputime = 616833506, rem_usecs = 18}, txend_arg = 0x0, 
  txend_cb = 0x0, phy_start_cputime = 616833416}
```

```
(gdb) p g_ble_ll_data
$2 = {ll_supp_features = 4110, ll_state = 0 '\000', ll_num_acl_pkts = 4 '\004', ll_acl_pkt_size = 255, ll_pref_tx_phys = 0 '\000', ll_pref_rx_phys = 0 '\000', ll_evq = {evq = {
      evq_owner = 0x200139d0 <g_ble_ll_task>, evq_task = 0x0, evq_list = {stqh_first = 0x20016b68 <g_ble_ll_rfmgmt_data+52>, stqh_last = 0x20016b74 <g_ble_ll_rfmgmt_data+64>}}}, ll_wfr_timer = {
    bsp_timer = 0x0, cb_func = 0x0, cb_arg = 0x0, expiry = 0, link = {tqe_next = 0x0, tqe_prev = 0x0}}, ll_rx_pkt_ev = {ev = {ev_queued = 0 '\000', ev_cb = 0x29831 <ble_ll_event_rx_pkt>, ev_arg = 0x0, 
      ev_next = {stqe_next = 0x20016b68 <g_ble_ll_rfmgmt_data+52>}}}, ll_rx_pkt_q = {stqh_first = 0x0, stqh_last = 0x20013790 <g_ble_ll_data+72>}, ll_tx_pkt_ev = {ev = {ev_queued = 0 '\000', 
      ev_cb = 0x297c9 <ble_ll_event_tx_pkt>, ev_arg = 0x0, ev_next = {stqe_next = 0x0}}}, ll_tx_pkt_q = {stqh_first = 0x0, stqh_last = 0x200137a8 <g_ble_ll_data+96>}, ll_dbuf_overflow_ev = {ev = {
      ev_queued = 0 '\000', ev_cb = 0x297c3 <ble_ll_event_dbuf_overflow>, ev_arg = 0x0, ev_next = {stqe_next = 0x0}}}, ll_comp_pkt_ev = {ev = {ev_queued = 0 '\000', 
      ev_cb = 0x297bd <ble_ll_event_comp_pkts>, ev_arg = 0x0, ev_next = {stqe_next = 0x0}}}, ll_hw_err_timer = {co = {c_ev = {ev_queued = 0 '\000', ev_cb = 0x298a5 <ble_ll_hw_err_timer_cb>, ev_arg = 0x0, 
        ev_next = {stqe_next = 0x0}}, c_evq = 0x20013758 <g_ble_ll_data+16>, c_ticks = 0, c_next = {tqe_next = 0x0, tqe_prev = 0x0}}}}
```

```
(gdb)  p g_ble_ll_scan_sm
$3 = {scan_enabled = 0 '\000', own_addr_type = 1 '\001', scan_filt_dups = 0 '\000', scan_rsp_pending = 0 '\000', scan_rsp_cons_fails = 1 '\001', scan_rsp_cons_ok = 0 '\000', 
  scan_peer_rpa = "\000\000\000\000\000", pdu_data = {hdr_byte = 195 '\303', {scana = "|\210\034\063", <incomplete sequence \303>, inita = "|\210\034\063", <incomplete sequence \303>}, 
    adva = "9~H\276\317", <incomplete sequence \327>}, upper_limit = 4, backoff_count = 3, scan_win_start_time = 0, scan_sched_ev = {ev = {ev_queued = 0 '\000', 
      ev_cb = 0x2fabf <ble_ll_scan_event_proc>, ev_arg = 0x20016db4 <g_ble_ll_scan_sm>, ev_next = {stqe_next = 0x0}}}, scan_timer = {bsp_timer = 0x2000c610 <nrf52_hal_timer5>, 
    cb_func = 0x2f925 <ble_ll_scan_timer_cb>, cb_arg = 0x20016db4 <g_ble_ll_scan_sm>, expiry = 616833867, link = {tqe_next = 0x20016f50 <g_ble_ll_sched_timer>, tqe_prev = 0x0}}, 
  scan_interrupted_ev = {ev = {ev_queued = 0 '\000', ev_cb = 0x30441 <ble_ll_scan_interrupted_event_cb>, ev_arg = 0x0, ev_next = {stqe_next = 0x0}}}, duration_timer = {
    bsp_timer = 0x2000c610 <nrf52_hal_timer5>, cb_func = 0x30149 <ble_ll_scan_duration_timer_cb>, cb_arg = 0x20016db4 <g_ble_ll_scan_sm>, expiry = 616833516, link = {
      tqe_next = 0x20016de8 <g_ble_ll_scan_sm+52>, tqe_prev = 0x0}}, period_timer = {bsp_timer = 0x2000c610 <nrf52_hal_timer5>, cb_func = 0x2fcc5 <ble_ll_scan_period_timer_cb>, 
    cb_arg = 0x20016db4 <g_ble_ll_scan_sm>, expiry = 0, link = {tqe_next = 0x0, tqe_prev = 0x0}}, duration_ticks = 51773, period_ticks = 0, ext_scanning = 0 '\000', 
[... cut ...]
      own_addr_type = 0 '\000', scan_filt_policy = 0 '\000', configured = 0 '\000', scan_type = 0 '\000', scan_chan = 0 '\000', timing = {interval = 0, window = 0, start_time = 0}}}}
```

```
(gdb) p p ble_ll_stats
No symbol "p" in current context.
(gdb) 
```

```
(gdb) p p g_ble_ll_conn_create_sm
No symbol "p" in current context.
(gdb) 
```

I will keep the DUT trapped in the debugger in this state. Please let me know if there is any other globals you would like me instead while its in this state to help you root cause the issue here.

andrzej-kaczmarek, 2020-06-29:
@lance-proxy thanks a lot. the problem seems to be triggered by ADV_EXT_IND received at the same time scan is disabled although I need to look through the code to figure out proper fix. could you please also dump `hdr` and `om->om_data[0]` parameters in frame 2 or 3?

andrzej-kaczmarek, 2020-06-29:
this should fix the problem (and some more...) https://github.com/apache/mynewt-nimble/pull/836

lance-proxy, 2020-06-30:
Many thanks @andrzej-kaczmarek! I will load your PR on the same test device, and monitor if this assertion occurs or not.

lance-proxy, 2020-06-30:
@andrzej-kaczmarek I see #836 is approved, so if we can get this merged to `master` today that would be great so we can do broader confidence testing of this patch.

andrzej-kaczmarek, 2020-06-30:
@lance-proxy I've just merged it

lance-proxy, 2020-07-01:
@andrzej-kaczmarek no sign of this assertion after updating our test device fleet with `master` (including your PR fix above). Many thanks again for the quick turnaround fix on this! Im going to close this issue unless the issue re-appears.

closed 2020-07-01 by lance-proxy