apache/mynewt-nimble #835: NimBLE Gap device name limits (I need BLE name with 25 chars).

adrianolmar, 2020-06-24:
Hi, I'm using NimBLE on espressif ESP32 IDF version 4.0.

I have problems with the size of name of device (ble_svc_gap_device_name_set). We can put until 19 characters on device name.
I know the advertise table is limited in 31 Octets.  logging adv table: these 31 bytes is 3 for flags, 3 for tx power, 4 for service uuid, 2 for name header ... it lasts 19 bytes for name as I imagine.

How we can use a additional "scan response" with more 31 bytes to solve my problem.

rymanluk, 2020-07-08:
Hi,

`ble_svc_gap_device_name_set` is used to set name for GAP DEVICE NAME characteristic in GAP service and not advertising i.e
Advertising data and scan response data you can configure independently.

adrianolmar, 2020-07-08:
How can we configure scan response? Do you have an example.

prasad-alatkar, 2020-07-13:
Hi @adrianolmar you can use `ble_gap_adv_rsp_set_fields` to set scan response. If you have ESP32 specific issues, feel free to raise issues here: https://github.com/espressif/esp-nimble/issues

rymanluk, 2020-07-15:
@adrianolmar I would look here: https://github.com/apache/mynewt-nimble/blob/master/apps/peripheral/src/main.c
and here:
https://github.com/apache/mynewt-nimble/blob/master/apps/ext_advertiser/src/main.c

KKopyscinski, 2021-05-20:
Because of no further replies, I'm assuming issue is solved, but feel free to reopen if needed @adrianolmar

closed 2021-05-20 by KKopyscinski