apache/mynewt-nimble #847: Porting NimBLE for CheriOS

hbcam, 2020-07-26:
I did a CheriOS port of NimBLE host controller.
I can successfully run the bleprph demo.

CheriOS runs on a CHERI-MIPS CPU ( https://www.cl.cam.ac.uk/research/security/ctsrd/cheri/ ), which has some additional programming constraints compared to your usual ARM/AMD46 CPU. I thus had to modify some non-npl files to make the port work.

I will open a few pull requests to see what can be merged upstream.

andrzej-kaczmarek, 2020-07-29:
would be great if you could add this port to mynewt-nimble repository - we can then run builds automatically on travis (same way as we do for e.g. RIOT OS) to make sure we do not break it in future.

hbcam, 2020-07-29:
The port-specific code is currently more a PoC than some nice code. I will need to clean it first.