apache/mynewt-nimble #858: Possible NULL dereference in porting/nimble/src/os_mempool.c

tluio, 2020-09-02:
On line 81, the pointer **membuf** is checked in 
  if (membuf != NULL)
This indicates that **membuf** can be NULL.
If so, a NULL dereference bug will occur on line 114:
  SLIST_NEXT(block_ptr, mb_next)

sjanc, 2020-09-04:
if membuf is NULL, blocks must be 0 (otherwise function returns in line 81) so that while() never loops and thus SLIST_NEXT(block_ptr, mb_next) is not called

tluio, 2020-09-04:
Sorry, but I wonder if our code is the same? If **blocks** is 0, SLIST_NEXT(block_ptr, mb_next) is also called in line 114.

sjanc, 2020-09-18:
Looks like this was already fixed in core but nimble copy wasn't updated. We should probably sync copy with core

KKopyscinski, 2021-05-20:
Fixed in merged patch

closed 2021-05-20 by KKopyscinski