apache/mynewt-nimble #864: Porting nimBLE to NuttX

v01d, 2020-09-16:
I'm experimenting with porting nimBLE to NuttX. We're in the process of completing a Linux-like socket interface via BTPROTO_HCI and BTPROTO_L2CAP. There's already a host layer on NuttX but I would like to try nimBLE as it already supports Linux and it shouldn't be much harder to support.

I'm finding some issues that I wanted to mention, not sure if these are possibles areas of improvement of nimBLE or I'm simply not fully understanding how to do the port (I couldn't find any documentation). 

For starters, I copied linux/ in npl/ and modified it to suit NuttX. As it is mostly Linux compatible, no big changes are needed. However one issue I find is that NuttX requires an initial include in every .c and .h as: `#include <nuttx/config.h>` which holds the output of the Kconfig configuration system. For the OS specific files in npl/ I can add this include, but for the rest of the code I would have to manually insert it everywhere. I imagine one way would be to extend the porting layer to have a header which would be included everywhere in nimBLE where I could add this include ([LVGL](https://github.com/littlevgl/lvgl) for example does it this way).

Another issue I find is there there are some macros which are not defined anywhere in nimBLE repo. For now I found (but I think there are many others):
* NIMBLE_CFG_CONTROLLER
* OS_CFG_ALIGN_4
* OS_CFG_ALIGNMENT
* BLE_LL_CFG_FEAT_LL_PRIVACY
* LOG_LEVEL_*

I'm sensing that there should be some configuration process for the port which defines these but it does not appear to be part of this repo (maybe done in mynewt repo?)

I also started by attempting to build the linux "test" applications but it failed due to `ble_npl_eventq_run` not being declared. Maybe this test is not maintained? If you don't think it is reasonable to spend time trying to build this test I can skip it. I was now moving on to trying to build the linux "example".

One final question: in Linux there's BTPROTO_L2CAP and on NuttX we're following the same path of exposing L2CAP over a socket. I see nimBLE only uses the HCI socket and solves L2CAP on its own. Is there any plan to make this optional and have BTPROTO_L2CAP sockets supported? This would not only better integrate nimble with NuttX but also Linux IMHO.

That's it so far. If you could give me some points on the porting process it would really be helpful.

v01d, 2020-09-16:
I think I'm understanding that i need to generate a syscfg.h for my platform. I've found some information in other issues here on how to supposeddly do this. I've created a newt project, changed version to 0.0.0, ran newt upgrade and then placed my modified porting files for nuttx in the nimble subdirectory. However, running `newt build @apache-mynewt-nimble/porting/targets/nuttx` produces the following output:
<pre>
* Warning: Parsing pkg @apache-mynewt-core/boot/split config: strconv.ParseInt: parsing "MYNEWT_VAL(SPLIT_APP_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/fs/fatfs config: strconv.ParseInt: parsing "MYNEWT_VAL(FATFS_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/fs/nffs config: strconv.ParseInt: parsing "MYNEWT_VAL(NFFS_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/battery config: strconv.ParseInt: parsing "MYNEWT_VAL(BATTERY_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/bsp/nordic_pca20020 config: strconv.ParseInt: parsing "MYNEWT_VAL(NORDIC_PCA20020_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/bsp/ruuvitag_rev_b config: strconv.ParseInt: parsing "MYNEWT_VAL(RUUVITAG_REV_B_SYSINIT_STAGE_BME280)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/bsp/stm32f3discovery config: strconv.ParseInt: parsing "MYNEWT_VAL(STM32F3DISCOVERY_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/charge-control config: strconv.ParseInt: parsing "MYNEWT_VAL(CHARGE_CONTROL_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/drivers/bq27z561 config: strconv.ParseInt: parsing "MYNEWT_VAL(BQ27Z561_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/drivers/ipc_nrf5340 config: strconv.ParseInt: parsing "MYNEWT_VAL(IPC_NRF5340_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/drivers/lwip/stm32_eth config: strconv.ParseInt: parsing "MYNEWT_VAL(STM32_ETH_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/drivers/pwm/soft_pwm config: strconv.ParseInt: parsing "MYNEWT_VAL(SOFT_PWM_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/drivers/uart/uart_bitbang config: strconv.ParseInt: parsing "MYNEWT_VAL(UARTBB_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/mcu/dialog/da1469x/hosttest config: strconv.ParseInt: parsing "MYNEWT_VAL(DA1469x_HOSTTEST_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/sensor/creator config: strconv.ParseInt: parsing "MYNEWT_VAL(SENSOR_CREATOR_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/sensor config: strconv.ParseInt: parsing "MYNEWT_VAL(SENSOR_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/usb/tinyusb/cdc_console config: strconv.ParseInt: parsing "MYNEWT_VAL(CONSOLE_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/hw/usb/tinyusb config: strconv.ParseInt: parsing "MYNEWT_VAL(USBD_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/kernel/os config: strconv.ParseInt: parsing "MYNEWT_VAL(OS_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/mgmt/imgmgr config: strconv.ParseInt: parsing "MYNEWT_VAL(IMGMGR_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/mgmt/oicmgr config: strconv.ParseInt: parsing "MYNEWT_VAL(OICMGR_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/mgmt/smp/transport/ble config: strconv.ParseInt: parsing "MYNEWT_VAL(SMP_BLE_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/mgmt/smp/transport/smp_shell config: strconv.ParseInt: parsing "MYNEWT_VAL(SMP_SHELL_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/mgmt/smp/transport/smp_uart config: strconv.ParseInt: parsing "MYNEWT_VAL(SMP_UART_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/mgmt/smp config: strconv.ParseInt: parsing "MYNEWT_VAL(SMP_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/net/ip/lwip_mn config: strconv.ParseInt: parsing "MYNEWT_VAL(IP_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/net/ip/native_sockets config: strconv.ParseInt: parsing "MYNEWT_VAL(NATIVE_SOCKETS_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/net/lora/node config: strconv.ParseInt: parsing "MYNEWT_VAL(LORA_NODE_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/net/oic config: strconv.ParseInt: parsing "MYNEWT_VAL(OC_SYSINIT_STAGE_TCP4)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-core/sys/config config: strconv.ParseInt: parsing "MYNEWT_VAL(CONFIG_SYSINIT_STAGE_2)": invalid syntax; ignoring package.
[... cut ...]
* Warning: Parsing pkg @apache-mynewt-nimble/nimble/transport/uart config: strconv.ParseInt: parsing "MYNEWT_VAL(BLE_TRANS_UART_SYSINIT_STAGE)": invalid syntax; ignoring package.
* Warning: Parsing pkg @apache-mynewt-nimble/nimble/transport/usb config: strconv.ParseInt: parsing "MYNEWT_VAL(BLE_TRANS_USB_SYSINIT_STAGE)": invalid syntax; ignoring package.
Building target @apache-mynewt-nimble/porting/targets/nuttx
Error: Could not resolve package dependency: @apache-mynewt-core/kernel/os; depender: porting/targets/nuttx
</pre>

Is it really necessary to go through this process to generate the syscfg.h? Or is there an easier way?

ccollins476ad, 2020-09-16:
@v01d I am not sure if there is a better way to generate a syscfg.h file.  The method you are using seems good to me.

Those warning messages look like your version of newt is old.  Which version are you using?  You can get the latest here: http://mynewt.apache.org/download/

v01d, 2020-09-16:
Thanks, yes, I've found some email in mailing list archive with similar error and it was suggested to use newt from master. This problem disappeared then. However, I'm struggling now with some undefined macros again. It seems that building nimble also tries to build mynewt OS which I would guess it is not necessary to generate the syscfg.h for a different OS:
<pre>
gcc -DAPP_NAME=dummy_app -DAPP_dummy_app -DARCH_NAME=dummy -DARCH_dummy -DBSP_NAME=dummy_bsp -DBSP_dummy_bsp -DMN_LINUX -DMYNEWT=1 -O0 -Wall -Werror -fdata-sections -ffunction-sections -fno-common -ggdb -m32 -Ibin/@apache-mynewt-nimble/porting/targets/nuttx/generated/include -Ibin/porting/targets/nuttx/user/pre_build/include -Irepos/apache-mynewt-core/crypto/tinycrypt/include -Irepos/apache-mynewt-core/crypto/tinycrypt/include/tinycrypt/arch/dummy -Irepos/apache-mynewt-core/hw/hal/include -Irepos/apache-mynewt-core/hw/hal/include/hal/arch/dummy -Irepos/apache-mynewt-core/kernel/os/include -Irepos/apache-mynewt-core/kernel/os/include/os/arch/dummy -Irepos/apache-mynewt-core/kernel/os/src -Irepos/apache-mynewt-core/kernel/os/src/arch/dummy -Irepos/apache-mynewt-core/sys/console/stub/include -Irepos/apache-mynewt-core/sys/console/stub/include/stub/arch/dummy -Irepos/apache-mynewt-core/sys/defs/include -Irepos/apache-mynewt-core/sys/defs/include/defs/arch/dummy -Irepos/apache-mynewt-core/sys/flash_map/include -Irepos/apache-mynewt-core/sys/flash_map/include/flash_map/arch/dummy -Irepos/apache-mynewt-core/sys/log/common/include -Irepos/apache-mynewt-core/sys/log/common/include/common/arch/dummy -Irepos/apache-mynewt-core/sys/log/modlog/include -Irepos/apache-mynewt-core/sys/log/modlog/include/modlog/arch/dummy -Irepos/apache-mynewt-core/sys/log/stub/include -Irepos/apache-mynewt-core/sys/log/stub/include/stub/arch/dummy -Irepos/apache-mynewt-core/sys/mfg/include -Irepos/apache-mynewt-core/sys/mfg/include/mfg/arch/dummy -Irepos/apache-mynewt-core/sys/stats/stub/include -Irepos/apache-mynewt-core/sys/stats/stub/include/stub/arch/dummy -Irepos/apache-mynewt-core/sys/sys/include -Irepos/apache-mynewt-core/sys/sys/include/sys/arch/dummy -Irepos/apache-mynewt-core/sys/sysdown/include -Irepos/apache-mynewt-core/sys/sysdown/include/sysdown/arch/dummy -Irepos/apache-mynewt-core/sys/sysinit/include -Irepos/apache-mynewt-core/sys/sysinit/include/sysinit/arch/dummy -Irepos/apache-mynewt-core/util/mem/include -Irepos/apache-mynewt-core/util/mem/include/mem/arch/dummy -Irepos/apache-mynewt-core/util/rwlock/include -Irepos/apache-mynewt-core/util/rwlock/include/rwlock/arch/dummy -Irepos/apache-mynewt-nimble/nimble/host/include -Irepos/apache-mynewt-nimble/nimble/host/include/host/arch/dummy -Irepos/apache-mynewt-nimble/nimble/host/services/gap/include -Irepos/apache-mynewt-nimble/nimble/host/services/gap/include/gap/arch/dummy -Irepos/apache-mynewt-nimble/nimble/host/services/gatt/include -Irepos/apache-mynewt-nimble/nimble/host/services/gatt/include/gatt/arch/dummy -Irepos/apache-mynewt-nimble/nimble/include -Irepos/apache-mynewt-nimble/nimble/include/nimble/arch/dummy -Irepos/apache-mynewt-nimble/nimble/transport/socket/include -Irepos/apache-mynewt-nimble/nimble/transport/socket/include/socket/arch/dummy -Irepos/apache-mynewt-nimble/porting/npl/mynewt/include -Irepos/apache-mynewt-nimble/porting/npl/mynewt/include/mynewt/arch/dummy -Irepos/apache-mynewt-nimble/porting/targets/dummy_app/include -Irepos/apache-mynewt-nimble/porting/targets/dummy_app/include/dummy_app/arch/dummy -Irepos/apache-mynewt-nimble/porting/targets/dummy_app/src -Irepos/apache-mynewt-nimble/porting/targets/dummy_app/src/arch/dummy -Irepos/apache-mynewt-nimble/porting/targets/dummy_bsp/include -Irepos/apache-mynewt-nimble/porting/targets/dummy_bsp/include/dummy_bsp/arch/dummy -Irepos/apache-mynewt-nimble/porting/targets/dummy_bsp/src -Irepos/apache-mynewt-nimble/porting/targets/dummy_bsp/src/arch/dummy -Irepos/apache-mynewt-nimble/porting/targets/nuttx/include -Irepos/apache-mynewt-nimble/porting/targets/nuttx/include/nuttx/arch/dummy -Irepos/apache-mynewt-nimble/porting/targets/nuttx/src -Irepos/apache-mynewt-nimble/porting/targets/nuttx/src/arch/dummy -MM -MG repos/apache-mynewt-core/kernel/os/src/os.c
Error: In file included from repos/apache-mynewt-core/kernel/os/include/os/os.h:131,
                 from repos/apache-mynewt-core/kernel/os/include/os/mynewt.h:27,
                 from repos/apache-mynewt-core/crypto/tinycrypt/src/mynewt.c:20:
repos/apache-mynewt-core/kernel/os/include/os/os_mempool.h:165:2: error: #error "Unhandled `OS_ALIGNMENT` for `os_membuf_t`"
  165 | #error "Unhandled `OS_ALIGNMENT` for `os_membuf_t`"
      |  ^~~~~
</pre>

I have my nimble_npl_os.h defining `#define BLE_NPL_OS_ALIGNMENT    4` which I think should work but it seems that it is not trying to use the nuttx porting headers here. In fact I see an os_mempool.h in both mynewt and the porting layer. How can I skip building the OS and just use the porting layer to generate syscfg?

ccollins476ad, 2020-09-16:
If the build got that far, the syscfg.h file should be generated.
```
bin/@apache-mynewt-nimble/porting/targets/nuttx/generated/include/syscfg/syscfg.h
```

v01d, 2020-09-16:
Ah, you're right. I see it there. Ok, I will continue then to build nimBLE from within NuttX using this syscfg.h.

v01d, 2020-09-16:
> However one issue I find is that NuttX requires an initial include in every .c and .h as: `#include <nuttx/config.h>` which holds the output of the Kconfig configuration system. For the OS specific [...]

Do you have any thoughts on how this could be addressed?

v01d, 2020-09-16:
I'm hitting an issue where some controller-related macros are not defined (not that I'm not using controller), even after adding them to syscfg.yml. For example, BLE_CONTROLLER, which gives this warning: `/home/v01d/coding/nuttx_nrf_ble/apps/external/nimble/mynewt-nimble/nimble/transport/socket/src/ble_hci_socket.c:793:5: warning: "MYNEWT" is not defined, evaluates to 0 [-Wundef]`. It builds anyway but it doesn't seem to be the right way to do it this way.

Also, note that I tried building `@apache-mynewt-nimble/porting/targets/linux` target and I get the same error as before for nuttx target so I'm guessing I'm hitting some issues present in that port as well.

v01d, 2020-09-19:
> Do you have any thoughts on how this could be addressed?

Nevermind about this, I realized it is not a problem.

Where I'm really stuck is with the undefined macros. Another problem is that some macros clash with NuttX's. Not sure how we could deal with those. The relevant errors are here: https://gist.github.com/v01d/342b63aee0f7a94a344f16de12a4defd#file-output-L1162
If we were to fix these I think we would be already in good place to have it working with NuttX (the rest of the changes we need to make on our side, getting the socket interface up to shape).

andrzej-kaczmarek, 2020-09-19:
`MYNEWY_VAL` macros are not defined because relevant syscfg settings are only defined when NimBLE Controller package is included in build. if you'd like to fix those warnings properly, some files would need extra ifdefs to skip declarations that are only used by controller, but I don't know if this will be possible for all warnings (did not check all of them tbh). another option is just to build NimBLE without `-Wundef` as we do in Mynewt :)

as for `LOG_LEVEL_*` warning, seems like log headers we have in NimBLE are a bit out of sync with mynewt-core, i.e. required definitions are in `log_common/log_common.h` file and it's included from `log/log.h`. copies in mynewt-nimble are missing this include so feel free to sync those headers with latest mynewt-core version (we unfortunately had to include copies of some files from mynewt-core to make build possible without that repo; long term solution would be to have more abstraction in NimBLE so it does not use Mynewt structures directly, but it requires quite a lot of work)

v01d, 2020-09-19:
Thanks for your response

> `MYNEWY_VAL` macros are not defined because relevant syscfg settings are only defined when NimBLE Controller package is included in build. if you'd like to fix those warnings properly, some files [...]

I could look into contributing some of this to nimBLE eventually. I will just ignore the undefined macros for now.

> as for `LOG_LEVEL_*` warning, seems like log headers we have in NimBLE are a bit out of sync with mynewt-core, i.e. required definitions are in `log_common/log_common.h` file and it's included from [...]

I understand. Keeping nimBLE standalone is a good choice. 
I copied the `log_common.h` file and removed an include which pointed to an OS file. I also added an include to `log_common.h` from `modlog.h`. With this, all LOG_LEVEL_* warnings disappeared =D

There one final problematic issue: LOG_* (such as LOG_DEBUG, etc) clash with those used in NuttX (see error here https://gist.github.com/v01d/342b63aee0f7a94a344f16de12a4defd#file-output-L1190). Since NuttX is POSIX compatible, these are clashing to standard definitions required for `syslog()` function (you can see these defined in `man syslog`). Maybe there's a way to add some prefix to these variables so that nimBLE can be built under a POSIX setting safely?

v01d, 2020-09-19:
Mmm, actually I just commented this definitions out in `log.h` so I guess they are not needed. Is that correct?

andrzej-kaczmarek, 2020-09-21:
They are probably not needed since they were just copied as-is from mynewt-core to make sure everything builds. However, I'd prefer not to make any modifications in those copied files so we can just update them with most recent copy from mynewy-core if needed without need to worry about some changes.

What you can do is to create simple headers with only the required stubs or macros for logging, have them in NuttX porting directory and set includes path to make sure they are used instead of our copied. This way you don't need to worry about contents of our log.h and related files.

v01d, 2020-09-21:
I understand. The problem is that `modlog.h` (another file under `porting/nimble/include`) does `#include "log/log.h"` so I cannot override it with include flags. `modlog.h` in turn is included from `host/include/host` headers.

v01d, 2020-09-21:
But `modlog.h` is also inside host/include and also uses `"` to include so I cannot override it. You can see the include path here:
```
In file included from /home/v01d/coding/nuttx_nrf_ble/apps/external/nimble/mynewt-nimble/porting/nimble/include/modlog/modlog.h:26,
                 from /home/v01d/coding/nuttx_nrf_ble/apps/external/nimble/mynewt-nimble/nimble/host/include/host/ble_hs_log.h:23,
                 from /home/v01d/coding/nuttx_nrf_ble/apps/external/nimble/mynewt-nimble/nimble/host/include/host/ble_hs.h:38,
                 from /home/v01d/coding/nuttx_nrf_ble/apps/external/nimble/mynewt-nimble/porting/examples/nuttx/ble.c:25:
/home/v01d/coding/nuttx_nrf_ble/apps/external/nimble/mynewt-nimble/porting/nimble/include/log/log.h:34: warning: "LOG_DEBUG" redefined
```

andrzej-kaczmarek, 2020-09-21:
Perhaps I am missing something here, but if you add e.g. `porting/nuttx/include/modlog/modlog.h` with stubs only and add `porting/nuttx/include` to include search path **before** `porting/nimble/include`, then your file should be always used instead of ours.

v01d, 2020-09-21:
I just tried that and it doesn't work. You cannot override include search paths for `#include` directives using `"` instead of `<`/`>`. If it were included as `#include <modlog/modlog.h>` you can indeed override it. I just tried changing that and that indeed works. But this would require modifying files under `nimble/host/include/host`. Maybe it is not a bad change to introduce as I suspect it will simplify this whole process of header copies.

andrzej-kaczmarek, 2020-09-24:
ok, makes sense :) so you can just go ahead and change `""` to `<>` in nimble if this solves your problem, we're fine with such change

v01d, 2020-09-24:
Cool! I'm hoping I can merge the required change on NuttX side and then I can open a PR here so we can discuss the changes on nimBLE side.

v01d, 2020-10-29:
Hi, you can find the PR here: #878

sjanc, 2021-05-27:
NuttX port is already merged

closed 2021-05-27 by sjanc