apache/mynewt-nimble #868: Power consumption during non-connectable advertising

mohit3112, 2020-10-02:
I am using a NRF51822 chip in DC/DC mode. 
with adv interval 150ms, 0 dbm tx power ,and BLE_LP_CLOCK 1. 
I am getting a current consumption of 1.2mA. 

I tried changing the BLE_LL_TX_PWR_DBM to -10 but i see no impact.
Do we have any power numbers for nimble + Nrf51/Nrf52 in different modes ?

haukepetersen, 2020-10-30:
Hej, as it happens, I just recorded some numbers for the `nrf52dk` running RIOT this morning. All measured with the Nordic power profiler kit directly attached to the devkit:

[image]

The plot shows 7 measurements:
- node IDLE (radio off)
- non-connectable advertisements with 3,15,31byte payload
- connectable advertisements with 3,15,31byte payload

Hope this gives you some orientation?!

vbelloir, 2021-02-28:
Hi @haukepetersen , thanks for those measurements!

have you improved then simce October '20?

If I play with nrf [online power estimator](https://devzone.nordicsemi.com/nordic/power/w/opp/2/online-power-profiler-for-ble), with a  200ms connectable, with 15 bytes payload, I get 58µA.
[image]

On your graph, it seems that it is  the green line, at about 350µA isn't it?

I know that this estimator uses SoftDevice instead of nimble.

sjanc, 2021-05-27:
Floor current is most likely due to other peripherals (UART, LEDs etc) being powered