apache/mynewt-nimble #877: No documentation about Data Length Extension

Avamander, 2020-10-21:
There's currently no documentation about Data Length Extension. 

It is silly to list a feature on the front page of documentation that isn't documented in any normal way.

haukepetersen, 2020-10-30:
Yes, it is indeed not easy to find. Take a look at `nimble/controller/syscfg.yml`:
```
    BLE_LL_CFG_FEAT_DATA_LEN_EXT:
        description: >
            This option enables/disables the data length update procedure in
            the controller. If enabled, the controller is allowed to change the
            size of tx/rx pdu's used in a connection. This option has only
            minor impact on code size and non on RAM.
        value: '1'
```
And from my own experience, looking at 
- `BLE_LL_MAX_PKT_SIZE`
- `BLE_LL_CONN_INIT_MAX_TX_BYTES`, 
- `BLE_LL_SUPP_MAX_RX_BYTES`
- `BLE_LL_SUPP_MAX_TX_BYTES`

will add to the picture and hopefully achieve the desired behavior?!

Avamander, 2020-10-30:
Will investigate and test out, thanks @haukepetersen.

sjanc, 2021-05-27:
how was your investigation? any chances you could send PR with documentation update?