apache/mynewt-nimble #879: nrf52840: unexpected PHY behavior?

haukepetersen, 2020-10-30:
While doing a sequence of energy measurements, I noticed some strange behavior on the `nrf52840dk` (`nrf52840`). While running a simple application that sends out legacy advertisements every 500ms, I can see CPU activity between two advertisement events where there should be none. This happens in non-deterministic intervals.

Trace A (ADV interval: 500ms, payload 3 byte, non-connectable, non-discoverable):
[image]

Trace B (ADV interval: 500ms, payload 3 byte, undirected-connectable, general discoverable)
[image]

Running the exact same application on the `nrf52dk` (`nrf52832`) does not show this behavior, on that board the trace looks as expected. So from that I think this might be a PHY related issue on the `nrf52840`?!

I also wonder if this could be related to #767?

Background infos:
- running NimBLE (dba097972d28701c10b9af2a48af7d924c9cfca3) @ RIOT (35b6ccedf31f10a5f8e4f97609ad5b10c28bdc34)
- the application simply calls `ble_gap_adv_start()` with the following params:
```
struct ble_gap_adv_params adv_params = {
        .conn_mode = BLE_GAP_CONN_MODE_NON, // or BLE_GAP_CONN_MODE_UND for trace B
        .disc_mode = BLE_GAP_DISC_MODE_NON,  // or BLE_GAP_DISC_MODE_GEN for trace B
        .itvl_min = BLE_GAP_ADV_ITVL_MS(500),
        .itvl_max = BLE_GAP_ADV_ITVL_MS(500),
    };
```
- No additional (user) threads 
- NimBLE is configured as stated in `porting/targets/riot/syscfg.yml` with the following changes:
```
CFLAGS += -DMYNEWT_VAL_BLE_LL_TX_PWR_DBM=0
CFLAGS += -DMYNEWT_VAL_BLE_LL_MAX_PKT_SIZE=251
CFLAGS += -DMYNEWT_VAL_BLE_LL_CFG_FEAT_DATA_LEN_EXT=1
CFLAGS += -DMYNEWT_VAL_BLE_L2CAP_COC_MPS=200
```

haukepetersen, 2020-10-30:
Forgot to mention: I can see this behavior (almost) independent of the advertising interval and payload size. I tested advertising intervals in [20, 50, 100, 200, 500, 1000, 2000]ms and payload sizes in [3, 15,31]byte - for all these configurations I can see occasional CPU (radio?) activity between some advertising events.

andrzej-kaczmarek, 2020-11-03:
We use the same phy for both nRF52832 and nRF52840 so it should behave pretty much the same. Also form LL pov both are exactly the same. If those are caused by radio, we should see that activity on debug pins so you can try to configure:
BLE_PHY_DBG_TIME_TXRXEN_READY_PIN
BLE_PHY_DBG_TIME_ADDRESS_END_PIN
BLE_PHY_DBG_TIME_WFR_PIN
and capture those pins on logic analyzer.

haukepetersen, 2020-11-03:
> We use the same phy for both nRF52832 and nRF52840 so it should behave pretty much the same

I know, also the RIOT code suporting that CPU is almost identical - thats what make this issue kind of strange...

> If those are caused by radio, we should see that activity on debug pins so you can try to configure:

Good idea. Will try this out and let you know what I find.

sjanc, 2021-05-27:
is this stil an issue?