apache/mynewt-nimble #884: HCI H:5 (3-wire) for BlueZ

daanpape, 2020-11-13:
Hello,

Is there any possibility to use this example https://mynewt.apache.org/latest/tutorials/ble/blehci_project.html with the 3wire (H:5) HCI protocol. This is for nRF52832 hardware connected to BlueZ without hardware flow control.

sjanc, 2020-11-16:
Hi,

Currently UART driver supports only H4 and thus requires HW flow control

daanpape, 2020-11-16:
@sjanc ok thank you. I know that H4 requires hw flow control. Is there any example or roadmap to expose H5 on the UART bus without hardware flow control (it is optional in the H5 protocol)

sjanc, 2020-11-16:
it was always as 'nice to have feature' and we had no plans add support for it unless someone step up and contribute it